Ultimaker/Cura #1394: Cura 2.4 (Windows 10 beta) crashes on start when a machine def file overrides machine_extruder_count

Thf772, 2017-01-31:
Hello,

I'm trying to set up Cura to work with my own printer on Win 10 x64. I need Sailfish/Makerbot-ish GCODE, and it seems only Cura 2.4 has Makerbot. The printer is a FlashForge Creator (a clone of the Makerbot Replicator Dual), which has 2 extruders.

After starting a test print, I found out that both extruders started heating if I only specified one extruder in Cura. I tried setting `machine_extruder_count` to 2 in the printer definition file, but doing so made Cura crash on startup. More preciesely, the splash screen shows "Loading Machines", then "Setting up scene" and then Cura crashes (splash disappears and process terminates).

For the record, here is my `ff_creator.def.json`:

    {
        "id": "ff_creator",
        "version": 2,
        "name": "FlashForge Creator",
        "inherits": "fdmprinter",
        "metadata": {
            "visible": true,
            "author": "Clubelek",
            "manufacturer": "FlashForge",
            "category": "Other",
            "file_formats": "text/x-gcode"
        },

        "overrides": {
            "material_diameter": { "default_value": 1.75 },
            "layer_height": { "default_value": 0.2 },
            "material_print_temperature": { "default_value": 200 },
            "machine_extruder_count": { "default_value": 1 },
            "machine_heated_bed": { "default_value": true },
            "machine_center_is_zero": { "default_value": true },
            "machine_height": { "default_value": 150 },
            "machine_gcode_flavor": { "default_value": "Makerbot" },
            "machine_depth": { "default_value": 150 },
            "machine_width": { "default_value": 227 },
            "machine_name": { "default_value": "FlashForge Creator" },
            "machine_extruder_count": { "default_value": 1 },
            "material_bed_temp_wait": { "default_value": false },
            "material_print_temp_wait": { "default_value": false },
            "material_print_temp_prepend": { "default_value": false },
            "material_bed_temp_prepend": { "default_value": false },
            "machine_start_gcode": { "default_value": "; -- START GCODE --\nM104 S0; disable unused extruder\nT0; set primary extruder\nM73 P0; enable show build progress\nM109 S{material_bed_temperature_layer_0} T0; heat bed up to first layer temperature\nM104 S{material_print_temperature_layer_0} T0; set nozzle heater to first layer temperature\nM103; disable RPM\nG21; set units to mm\nG162 X Y F6000; home XY axes maximum\nG161 Z F9000; home Z axis minimum\nG92 Z0; set Z to 0\nM132 X Y Z A B; Recall stored home offsets\nG90; set positioning to absolute\nG1 Z30; move Z to waiting height\nG1 X-95 Y-73 Z30 F14000; move to waiting position (front left corner of print bed)\nG130 X0 Y0 A0 B0; set stepper motor vref to lower value while heating\nM6 T0; wait for bed and extruder to heat up\nG130 X127 Y127 A127 B127; set stepper motor vref to defaults\nM108 T0 R3; set extruder speed\nG92 E0; set E to 0\nG90; use absolute coordinates\nM320; acceleration enabled for all commands that follow\nG1 Z{layer_height_0} F6000.000; move to first layer height\nG1 X100 Y-73 F14000.000; move to front right corner of bed\nG1 X-90 Y-73 E24 F2000.000; extrude a line of filament across the front edge of the bed\nG4 P2000; wait for ooze to slow\nG1 Z0 F6000.000; lower nozzle height to 0\nG1 X-95; wipe nozzle\nG1 Z{layer_height_0} F6000.000; set nozzle to first layer height\nG1 F12000; ensure fast travel to first print move\nG92 E0; set E to 0 again\nM73 P0; reset build progress to 0\n; -- end of START GCODE --" },
            "machine_end_gcode": { "default_value": "; -- END GCODE --\nM73 P100 ; end build progress\nM109 S0 T0; set bed temperature to 0\nM104 S0 T0; set extruder temperature to 0\nM103; Disable extruder\nG1 Z150 F9000; lower plate\nG162 X Y  F2500; home X and Y axes\nG92 Z0; set Z to 0\nG1 X-10 Y-10 Z0 F9000; move print head away from endstops\nM18; disable all stepper motors\nM72 P1; Play Ta-Da song\n; -- end of END GCODE --" }
        }
    }

As you can see, I use a workaround which is to start the custom GCODE with `M104 S0; disable unused extruder` to shut down all heating. Overriding `material_print_temp_prepend` and `material_bed_temp_prepend` did nothing, Cura always prepends the start GCODE with M104 and M140 commands, which are not even the right ones for my printer.

fieldOfView, 2017-01-31:
Changing the number of extruders in a definition is a bit of work at the moment. See ultimaker_original_dual.def.json for an example. You have to specify the ```machine_extruder_trains``` metadata and provide the files (again, see ultimaker_original_dual_1st.def.json and ultimaker_original_dual_2nd.def.json, both are in the extruders folder).

Thf772, 2017-02-01:
Thanks, I'll look at it as soon as I have time. I'll post more when I test it.

Ghostkeeper, 2017-02-01:
I wouldn't expect it to crash though. Just to... not add the printer definition. For the crash we'd need the application log to begin debugging it: C:\Users\Thf772\AppData\Local\cura\stderr.log

CalebPetersenPhD, 2017-02-20:
Hey Thf772, thanks for trying to get the FF working in Cura Beta 2.  Did you have any luck, specifically with dual extrusion?  I'm currently using Simplify3D for dual extrusion (dissolvable support) on my FlashForge but am underwhelmed (e.g, even unlike crappy Slic3r, Simplify3D can't make just an interface out of dissolvable support material to save on its cost) and was hoping Cura 2 would have some nice new features (dual extrusion didn't work in Cura 1 on the FF, nor did the workarounds)

nallath, 2017-07-27:
This should already be fixed.

closed 2017-07-27 by nallath