Ultimaker/Cura #1396: spiralize not working right in 2.4

flogger8, 2017-01-31:
2.4
Mac
Set to spiralize, walls 1, infill 0, top 0, bottom 3 layers.  appears to spiralize, but still creates this code at the end and beginning of each layer 

G1 X80.749 Y80.717 Z3 E1286.47333
G0 F3600 X80.785 Y80.521
G0 X80.785 Y80.521 Z2.9
;TIME_ELAPSED:1893.177985
;LAYER:27
G0 X80.793 Y80.695 Z3
;TYPE:WALL-OUTER
G1 F1800 X80.848 Y80.407 E1286.4782

flogger8, 2017-01-31:
bigger gcode snippet

G1 X77.409 Y82.357 Z2.898 E1282.97698
G1 X77.536 Y82.218 E1282.98011
G1 X77.724 Y82.053 E1282.98427
G1 X77.892 Y81.954 E1282.98751
G1 X78.222 Y81.919 E1282.99303
G1 X78.356 Y81.892 E1282.99531
G1 X78.755 Y81.853 E1283.00197
G1 X78.811 Y81.839 E1283.00293
G1 X79.273 Y81.791 Z2.899 E1283.01066
G1 X79.565 Y81.768 E1283.01553
G1 X79.774 Y81.735 E1283.01905
G1 X79.802 Y81.668 E1283.02025
G1 X79.831 Y81.652 E1283.0208
G1 X80.067 Y81.418 E1283.02633
G1 X80.155 Y81.347 E1283.02821
G1 X80.33 Y81.154 E1283.03254
G1 X80.472 Y80.981 E1283.03627
G1 X80.639 Y80.758 E1283.0409
G1 X80.702 Y80.739 Z2.9 E1283.04199
G0 F3600 X80.74 Y80.543
G0 X80.74 Y80.543 Z2.8
;TIME_ELAPSED:1886.089495
;LAYER:26
G0 X80.749 Y80.717 Z2.9
;TYPE:WALL-OUTER
G1 F1800 X80.868 Y80.069 E1283.05295
G1 X80.931 Y79.837 E1283.05695
G1 X80.967 Y79.535 E1283.06201
G1 X81.03 Y79.311 E1283.06588
G1 X81.069 Y79.007 E1283.07097
G1 X81.169 Y78.516 Z2.901 E1283.07931
G1 X81.231 Y78.398 E1283.08152

BerndJM, 2017-01-31:
Seems the same I reported here few days ago:
https://ultimaker.com/en/community/37350-24-beta-what-happens-to-the-spiralize-feature

flogger8, 2017-01-31:
Yep… I saw your post and wasn't sure whether the developers needed a bug request posted to github or if they were following the community so I figured I'd post and let them delete if it was duplicate

Ghostkeeper, 2017-02-01:
> wasn't sure whether the developers needed a bug request posted to github or if they were following the community

It's mostly just Nallath and Daid that follow the forums as far as I know. A bug report down here never hurt anyone.

BagelOrb, 2017-02-01:
Most of our developers aren't very active on the forum. It's best to report issues on github.

smartavionics, 2017-02-02:
Just submitted a fix for a bug in the CuraEngine spiralize code, maybe that was the cause of your problem?

ntoff, 2017-02-18:
I just noticed the same behaviour on Cura2.4 Beta2 on windows and was about to submit a bug report but looks like this might be the same thing I have. It's almost like reverse z hop only z hop is turned off. 

Every layer change the nozzle goes down and digs into the print, seemingly retracts, and causes a big hole where a seam would usually be, only it's in spiralize mode and retract on layer change is off, so is z hop.

@smartavionics is there a way to test your bug fix?

smartavionics, 2017-02-18:
Hi, that fix has been merged into the master and 2.4 branches. If you are using a Linux host it should now be in the PPA packages. I don't know whether it has made it into any of the other testing releases.
Also, I do not know whether it will even solve the problem you are seeing. Cheers, mb

BagelOrb, 2017-02-18:
I wish I could solve this for the 2.4 release, but I'm afraid the final 2.4 release will have this bug.
It will definitely be fixed for 2.5, though.

ntoff, 2017-02-19:
>I wish I could solve this for the 2.4 release, but I'm afraid the final 2.4 release will have this bug.
It will definitely be fixed for 2.5, though.

Perhaps disable the spiralize feature for 2.4 then? Just comment out that checkbox? I mean I "wasted" an hour on the print http://imgur.com/a/o8GNf though it did turn out, the "seam" (which shouldn't even exist in spiral mode?) is pretty lumpy.

>Also, I do not know whether it will even solve the problem you are seeing. Cheers, mb

@smartavionics This is the behaviour I'm seeing https://www.youtube.com/watch?v=g2LN7GX_kT8 Not sure if it's going to be totally clear on video what's going on but in the "slow motion" part of it, you can clearly see the nozzle move down and back up where the "seam" is. Watch out for flickering lights, the 120fps mode on my phone doesn't seem to agree with my 50hz mains power.

ntoff, 2017-02-19:
I think I found a workaround, if you disable "outer wall wipe distance" (set it to 0) the gcode seems to be missing the weird dip code. The distance of the dip also corresponded to the wipe distance (0.2mm). I've not had time to comprehensively test it (currently midnight here) but maybe a quick fix (if it works) is to just set the wipe distance to 0 when spiral mode is on?

[WS_vase.txt](https://github.com/Ultimaker/Cura/files/785905/WS_vase.txt)

[WS_vase_nowipe.txt](https://github.com/Ultimaker/Cura/files/785907/WS_vase_nowipe.txt)

smartavionics, 2017-02-19:
I'm looking at the code now and what you say certainly has an influence. However on my little test object I am still seeing the weird behaviour I have seen before where it goes twice around the wall in a nice spiral for two turns and then it drops down and does another two turns in the other direction and then it jumps back up and does two turns in the original direction (and just carries on like that).

I will continue looking...

ntoff, 2017-02-19:
spiralize doesn't seem to affect shell count, I've manually set that to 1 shell and in my previous test print it did do only one shell, dip, up, 1 shell, dip, up, 1 shell, etc. Try manually setting the shell count to 1.

smartavionics, 2017-02-19:
I have just submitted a PR which for me, makes spiralize work (at least for a trivial example). Fingers crossed that it fixes the problems you are seeing and that it gets into 2.4.

smartavionics, 2017-02-20:
Sorry, the cura devs didn't like my PR so I have reverted it and have given up working on this as they have their own thoughts as to what needs to be done.

fieldOfView, 2017-02-20:
Aww, don't give up hope. Note that every contribution goes through a review process and changes are requested quite often. This is true even for the UM cura devs.

It was great seeing you two go "at it" this weekend. Unfortunately the build that will be released as the final version of Cura 2.4 has already been made (I think it is more than a week old by now).

smartavionics, 2017-02-20:
The problem here, I think, is that I seem to have the wrong idea as to how spiralize is meant to work. I think i'll stick to stuff I understand.

smartavionics, 2017-02-20:
I already have several Cura/CuraEngine PR's stacked up waiting for you guys to accept contributions again. I'm more interested in getting those moving.

nallath, 2017-02-20:
Sorry bout all that :(

I only merge Uranium & Cura "only" PR's.

BagelOrb, 2017-02-20:
Please see https://github.com/Ultimaker/CuraEngine/pull/465
I think there was some miscommunication about the PR.

Ghostkeeper, 2017-02-22:
As far as I know, pull requests on CuraEngine are still on hold pending the license switch. When that is going to take place though, I don't know.

Most of them are from Smartavionics. I'm keeping the notifications from Github in a to-do slot so they won't get forgotten.

BagelOrb, 2017-03-08:
I think this is fixed in 2.5 by the fixes from @smartavionics .

closed 2017-03-08 by BagelOrb

ntoff, 2017-03-09:
is 2.5 available anywhere as a beta? or do we need to compile it ourselves / run it from source (is that possible?)?

smartavionics, 2017-03-09:
Hi, I don't think any betas will be available quite yet. You can compile it yourself as all the source is available. Doing that is reasonably straightforward on Linux. Sorry, don't know about the other platforms.

BagelOrb, 2017-03-09:
Beta will hopefully be out somewhere next week.