Ultimaker/Cura #1397: Estimated time left stops counting down

KarelTenkink, 2017-02-01:
After an half an hour or so, the estimated time left counter on MACos CURA 2.4 beta surrounding stops counting down.

When CURA 2.4 is restarted it is solved

nallath, 2017-02-02:
What count down timer? The one that you see when connected with the UM3?

fieldOfView, 2017-02-02:
When it stops, could you check if the monitor icon in the top right of the window has a cyan "..." or grey "?" (or other color) overlay?

Ghostkeeper, 2017-02-02:
I suspect it might just be a lost connection. Cura should try to restore the connection on its own though, but perhaps it takes a while before it tries the connection again.

KarelTenkink, 2017-02-02:
The camera is still streaming, so in my opinion its not a lost connection.
[image]

KarelTenkink, 2017-02-02:
It has been running for more than 6 hours now. Estimated time left stopped at 3 hours into printing.

fieldOfView, 2017-02-02:
It is perfectly possible to have a working camera stream but no connection from Cura to the part of the firmware that supplies the printer status; they are two separate systems, and they can "break" independently. If the screenshot you posted is of the situation you describe (estimated time left stops going down), then Cura has not noticed it (yet).

KarelTenkink, 2017-02-02:
Ok thanks for replying.

fieldOfView, 2017-02-02:
Please keep the issue open; it needs to be investigated and fixed.

Ghostkeeper, 2017-06-05:
Apparently not. Our product manager just closed this because it was inactive for 3 months.

ianpaschal, 2018-02-27:
This is no longer an issue in Cura 3.2.1.

closed 2018-02-27 by ianpaschal