Ultimaker/Cura #1400: Material retraction distance and speed is NaN

fbertram, 2017-02-02:
When opening the print settings for material, the retraction distance and retraction speed are shown as NaN (for generic PLA). When looking at the material settings in the custom print setup, these fields actually have a value that seems to be loaded from the material profile (6.5mm and 25mm/s for generic PLA).

thopiekar, 2017-02-03:
Which printer are you using? Does it always occur?

fbertram, 2017-02-05:
I am using a Kodama Trinus 3D printer. However, I would be surprised to learn that this issue is related to the printer.

The issue is 100% reproducible w/ Cura 2.4 beta and it seems that all materials show NaN in Preferences/Materials. As mentioned above, in the material section on the main screen, these values are shown correctly.

Ghostkeeper, 2017-02-06:
Can't reproduce this:

[image]

The same shows up when using a different printer: Custom FDM Printer. It shows up as 0. This is because the material doesn't define a retraction distance or speed. I'd say NaN is not user-friendly at all so if that does show up in some case it needs to be changed. Showing 0 is still a bit unfriendly, but a more acceptable behaviour.

Question: Does it show up correctly for Nylon? Nylon has a speed and distance defined.

fieldOfView, 2017-02-06:
I can reproduce it:
[image]

Ghostkeeper, 2017-02-06:
Also for UM2+ it doesn't reproduce for me :(

Ghostkeeper, 2017-02-06:
Ah, it seems we have an issue for this in our tracker. It's only reproduced on Windows so far.

maukcc, 2017-02-08:
Also: Retraction speed and distance is not of any concern per material, it should be a machine specific amount.
Prime speed however is a material specific setting. 

The best setting would be prime speed for the last 1mm of priming. 
For instance PETG (which is notorious in priming) can be retracted 3mm@40mm/sec, then prime 2mm@40mm/sec and then prime 1mm@8mm/sec

Ghostkeeper, 2017-02-09:
Retraction speed and distance is certainly a concern per material. Specifically, flexible materials need to retract much farther than hard materials. It is also a concern per machine, since you'd like to retract fast enough to create under-pressure on the nozzle so that it won't drip, and you want to retract far enough for the same reason. But we can have the setting in both the machine and material. If it's specified in the material that will override the setting in the machine, but in the material it can be specified per machine.

We'd like to add more settings to these files, such as prime speed. But the entire XML format thing is something we're not quite happy with altogether. This is one of the reasons for it.

nallath, 2017-07-27:
Can't reproduce this.

closed 2017-07-27 by nallath