Ultimaker/Cura #1401: 295 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] [WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.

willyt, 2017-02-02:
I'm having trouble defining a dual extruder printer. CuraEngine seems to recognise that the printer has two extruders and generates gcode for both, but the gcode heats the second extruder even when it is not used. In the log I get a couple of messages that make me think I don't have it set up right. Like:
`2017-02-02 16:37:44,852 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [375]: Tried to get the extruder trains for machine k8400 Dual Extruder, which doesn't exist.` 
where it seems the code in MachineSettingsAction.py is trying to lookup the definition using the`name:` property rather than the `id:` property and later
`2017-02-02 16:56:54,295 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] [WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.`
 I've read the code that logs this message but it's beyond me to work out why extruder_train_ids.size() would be 1 if its based on the json in my k8400_dual.def.json file.
I'm wondering does Cura or CuraEngine emit heat up code for all extruders as a failsafe even if the 2nd extruder is never used? Also, why doesn't the 'Turn off nozzle after last use' feature kick in? I'd like to submit a pull request so you can support this printer, but I don't want to do that before I have resolved this issue as it could catch people out and cause problems.
I have attached my machine definition and my extruder def files.
[k8400_dual.def.json.txt](https://github.com/Ultimaker/Cura/files/748478/k8400_dual.def.json.txt)
[k8400_dual_1.def.json.txt](https://github.com/Ultimaker/Cura/files/748481/k8400_dual_1.def.json.txt)
[k8400_dual_2.def.json.txt](https://github.com/Ultimaker/Cura/files/748483/k8400_dual_2.def.json.txt)
[cura.log.txt](https://github.com/Ultimaker/Cura/files/748511/cura.log.txt)
I was trying to build Cura so I could fix it myself, but I'm afraid the build config just got a bit too complicated and I had to give up at Uranium, but let me know if I can help in any other way.

Best,
W

Ghostkeeper, 2017-02-06:
> 2017-02-02 16:37:44,852 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [375]: Tried to get the extruder trains for machine k8400 Dual Extruder, which doesn't exist.

This seems to me to be a mistake in your profiles, or otherwise a bug in Cura.

> 2017-02-02 16:56:54,295 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] [WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.

We get this error all the time from the engine. It doesn't seem to cause issues for the Ultimaker 3, but something is up there.

> I'm wondering does Cura or CuraEngine emit heat up code for all extruders as a failsafe even if the 2nd extruder is never used?

It shouldn't heat up unused extruders! You should be able to test if Cura thinks there are two extruders in this print if you have skirt or brim activated, because CuraEngine draws a line of skirt around the original brim/skirt with every extruder. If it doesn't draw a skirt or brim but does heat up the nozzle, that would seem like a bug.

> Also, why doesn't the 'Turn off nozzle after last use' feature kick in?

Ah, I thought we removed that. Did we leave the setting in? I can't find it in fdmprinter.def.json at master at the moment. The reason we removed it was because the UM3 would then still require heating up the nozzle at the end of the print in order to put the filament in a cold spot after the print (parking stance). We could retract the filament there just before cooling down, but that would leave a blob of material degrading in the nozzle. We should actually retract in two parts: First a bit to let the material cool down and harden a bit, then a bit more to pull the hardened bit of material out of the nozzle. It's complex.

willyt, 2017-02-06:
Here is the gcode file that is generated.
[KDE_JennyCorner.gcode.txt](https://github.com/Ultimaker/Cura/files/755133/KDE_JennyCorner.gcode.txt) Because `material_print_temp_prepend` is set to true, on lines 7 and 9 M104 and M109 heat up instructions are issued for the second extruder. Then there are a bunch of priming instructions from `machine_start_gcode`, in which line 21 switches back to the first extruder. After this no gcode is emitted which changes the tool until the `machine_end_gcode` where on line 59368 the second extruder is switched off having remained unused but kept at 215 ºC for the duration of the print, which would be about 2 hours in this case.

Ghostkeeper, 2017-02-07:
Sorry, I had intended to take a better look at your profiles but I was at home when I reacted previous time and didn't have the good tools I have at work to test it.

I spotted a few mistakes looking at your definition: There are `"value"` properties that are not strings. The `"default_value"` property should have a proper JSON type for the setting (like integer or boolean) but the `"value"` property is interpreted as a string of Python code. So all those numbers you have after the value need to be encased in quotes and the value of `retraction_hop_enabled` needs to be capitalised because of Python: `"True"`.

Some settings, such as Print Sequence's enabled property, are superfluous because it is already the default. That is not a real problem though, just bad form.

The printer definition is also in the category "Custom". If you're using this for yourself that's fine, but if you're intending to get this merged in the official Cura release, we'll need to change this to "Other".

When I test this, everything seems to be as you say: The front-end recognises the dual extrusion printer and seems to work fine. But when you slice it it will always warm the second nozzle. It seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/1128 , that one. An actual bug, that is, but a duplicate.

I checked it and indeed we removed the "Turn off nozzle after last use". The reason is as I described before: There is trouble with retracting the material to the correct position before cooling the nozzle. If done incorrectly this can ruin the feeder when removing the material (if the nozzle volume gets dragged along with the material in a loose ball) or ruin the nozzle (if the nozzle volume detaches from the filament entirely and can't be removed when removing the material). To implement it correctly with a two-stage retraction turned out to be quite difficult.

So to summarise: Your definition needs some minor fixing but is mostly okay. You also found a bug, which is an actual bug but a duplicate here on Github. You also miss a feature that was removed.

Ghostkeeper, 2017-02-07:
Here is a fixed version of the JSON you provided:

    {
        "id": "k8400_dual",
        "version": 2,
        "name": "k8400 Dual Extruder",
        "inherits": "fdmprinter",
        "metadata": {
            "visible": true,
            "author": "Will",
            "manufacturer": "Velleman",
            "category": "Other",
            "file_formats": "text/x-gcode",
            "has_materials": true,
            "machine_extruder_trains":
            {
                "0": "k8400_dual_1",
                "1": "k8400_dual_2"
            },
            "first_start_actions": ["MachineSettingsAction"]
        },
    
        "overrides": {
            "machine_name": { "default_value": "k8400" },
            "machine_width": { "default_value": 240 },
            "machine_height": { "default_value": 190 },
            "machine_depth": { "default_value": 215 },
            "machine_center_is_zero": { "default_value": false },
            "machine_nozzle_size": { "default_value": 0.35 },
            "machine_nozzle_heat_up_speed": { "default_value": 2 },
            "machine_nozzle_cool_down_speed": { "default_value": 2 },
            "machine_head_with_fans_polygon": { "default_value": [ [0, 200], [0, -20], [200, 200], [200, -20] ]},
            "gantry_height": { "default_value": 31 },
            "machine_use_extruder_offset_to_offset_coords": { "default_value": true },
            "machine_gcode_flavor": { "default_value": "RepRap (Marlin/Sprinter)" },
            
            "machine_start_gcode": {
                "default_value": "G21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM106 S165 ;start with the fan on\nG28\nG1 Z15.0 F{speed_travel} ;move the platform down 15mm\nT1 ;Switch to the 2nd extruder\nG92 E0 ;zero the extruded length\nG1 F100 E10 ;extrude 10 mm of feed stock\nG92 E0 ;zero the extruded length again\nG1 F100 E10\nT0 ;Switch to the 1st extruder\nG92 E0 ;zero the extruded length\nG1 F200 E6 ;extrude 6 mm of feed stock\nG92 E0 ;zero the extruded length again\nG1 F9000\n;Put printing message on LCD screen\nM117 Printing..."
            },
            "machine_end_gcode": {
                "default_value": "M104 T0 S0 ;1st extruder heater off\nM104 T1 S0 ;2nd extruder heater off\nM140 S0 ;heated bed heater off (if you have it)\nG91 ;relative positioning\nG1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure\nG1 Z+0.5 E-5 X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more\nG28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way\nG90 ;absolute positioning\nG1 Z190 F{machine_max_feedrate_z} \nM84 ;steppers off"
            },
            
            "machine_extruder_count": { "default_value": 2 },
            "print_sequence": { "enabled": false },
            "prime_tower_position_x": {"value": "185" },
            "prime_tower_position_y": { "value": "160" },
            "machine_max_feedrate_x": { "value": "160" },
            "machine_max_feedrate_y": { "value": "160" },
            "machine_max_feedrate_z": { "value": "10" },
            "material_diameter": { "value": "1.75" },
            "speed_print": { "value": "42" },
            "speed_travel": { "value": "180" },
            "material_flow": { "value": "90" },
            "cool_min_layer_time": { "default_value": 10 },
            "wall_thickness": { "value": "0.7" },
            "line_width": { "value": "0.35" },
            "retraction_hop_enabled": { "value": "True" }
            
        }
    }

BagelOrb, 2017-07-23:
I suppose the last post fixed the problem

closed 2017-07-23 by BagelOrb