Ultimaker/Cura #1402: Cura 2.x always opens a new instance when exporting from Fusion 360

godefroi, 2017-02-03:
When exporting a body or component as an STL, Fusion 360 allows the user to choose a 3D printing utility to export to. One of those utilities that can be exported to is Cura. When I first started working with Cura, in the 15.x days, choosing Cura as the export target would cause Cura to be launched with the STL loaded (if it was not already running) or would load the STL in the running instance (if an instance was already running). This was an extremely convenient workflow.

With Cura 2.x, however, the workflow is not nearly as convenient. Cura 2.x has to be set up as a "Custom" application in Fusion 360, and I assume that this means that Cura is simply started by calling the executable and passing a filename on the command line that refers to a temporary STL file. This results in a new Cura instance being started each time the STL is exported, whether or not an instance is already running.

If it is not possible, for whatever reason, to return to the integration system used by the older version of Cura, then I believe the next-best option is to add a command-line flag that would cause a newly-started Cura instance to look for an already-running instance, and if it exists, to pass the STL on to the already-running instance.

fieldOfView, 2017-02-03:
Note that if you select Autodesk's own MeshMixer for slicing, it behaves the exact same way; a new instance of MeshMixer is opened every time you press "Ok" in the Make dialog. So this may be by design by AutoDesk

godefroi, 2017-02-03:
I'm not sure it's "by design" more than it's a byproduct of how the software is launched. Clearly, the old version of Cura had the ability to elegantly handle this situation. I believe the Ultimaker team has been in contact with the Fusion 360 team, they may be able to provide some insight.

nallath, 2017-02-06:
They stopped responding after one or two messages. So no luck there.

awhiemstra, 2017-02-06:
Note that @sedwards2009 is currently working on a solution for this. This is a bit more tricky because we do not simply want to create a single-instance Cura, we have enough problems with the Python GIL as it is.

sedwards2009, 2017-02-08:
On Cura's git master I recently added a flag `--single-instance` which gives this behaviour. A CAD program should pass this flag to Cura when sending a model to it.

iFreilicht, 2017-10-15:
@sedwards2009 it seems like the `--single-instance` flag has been removed again. Using Cura 2.7.

I'd like to voice support for it to be added again. It's just stupid that I have to save my .STL from Fusion 360 to a temp folder and then manually open it in Cura. As a workaround, after opening it once, one can use F5 to update the model after exporting a new version from Fusion 360.

nallath, 2017-10-16:
The --single-instance flag is still there. It's up to the program that calls Cura to actually use the flag.

iFreilicht, 2017-10-16:
Are you 100% certain of that? I tried it on windows yesterday by entering the Cura install directory on the command line and executing `Cura.exe --single-instance` twice. The second execution opened a new window.

nallath, 2017-10-16:
Yup. Just tried it when running from source (on Fedora). Could be that it's broken for Windows though.

Ghostkeeper, 2017-10-22:
Nope, it still works on Windows 8.1 as well (for me), using Cura 3.0.3.

iFreilicht, 2017-10-23:
Huh, it might be a problem with Win10 or Powershell then. I'll try it out with cmd as well.

And I was not aware that Cura 3 was already released, maybe that's worth a shot.

vlachoudis, 2018-02-07:
After Cura V3.x fusion 360 doesn't find it as an installed 3d printer. A workaround is to create a ```cura.bat``` file with
```
@echo off
start "Cura" "C:\Program Files\Ultimaker Cura 3.2\Cura.exe" "%1" --single-instance
```
and set it as a custom 3d printer in fusion 360.

godefroi, 2018-02-07:
@vlachoudis there's a separate issue for that, and IIRC, the integration was a collaboration between the F360 people and the Cura people, and the F360 people haven't been eager to catch up.

closed 2018-02-27 by ianpaschal

Nickduino, 2018-06-15:
> After Cura V3.x fusion 360 doesn't find it as an installed 3d printer. A workaround is to create a cura.bat file [...] and set it as a custom 3d printer in fusion 360.

My Fusion 360 won't accept a *.bat (or a shortcut to a *.bat) as a custom printer

nukadelic, 2019-01-23:
@Nickduino same here

Nickduino, 2019-01-23:
In fact, it does :-)

It won't let you click on a *.bat but is ok with it if you type the file name.

TheModernWizard, 2019-12-02:
Where is the option to input a file name? I cannot find it. The only option I have it to click on the file in file explorer, and of course, .bat files don't show up in that window.

Ghostkeeper, 2019-12-05:
I think they're talking about when you call the .bat file from a command line.

TheModernWizard, 2019-12-05:
@Ghostkeeper Yes, that makes sense. Actually, for anyone still interested in this issue, contact Jerome Briot on the Fusion 360 Autodesk forums, he developed an add-in called "3d Print Plus", which solved this problem for me. It integrates the single instance flag into the export, so all you have to do is select Ultimaker Cura and export, and it opens it in your already existing instance of Cura. (or opens a new one if one does not yet exist)