Ultimaker/Cura #1403: Removable drive issues

fbertram, 2017-02-05:
observed with 2.4 beta:

when seeing a removable drive, Cura jumps right at it in order to make it convenient to save files to an SD card and this is sure an appreciated feature.

Unfortunately, Cura then seems to look for that drive on the next startup - and when that drive is no longer attached, there is no way to start Cura.

nallath, 2017-02-06:
OS? Logs?

awhiemstra, 2017-02-06:
This is a known issue with some SD card readers on Windows that incorrectly report they have a drive but actually do not. We then try to access the drive information from that drive which results in a stupid "abort, retry, cancel" popup from Windows. I do not currently know if this is even solvable.

Ghostkeeper, 2017-02-06:
Is there a way that we can survive the pop-up but just get an exception in Python? That would've been nice.

awhiemstra, 2017-02-06:
Not that I know of. It is basically the Windows driver giving up.

fbertram, 2017-02-06:
This is 2.4.0 beta 2 on Windows 10. Please note that this is not an issue with the SD card reader. Cura still won't start with the card reader disconnected.

As a workaround, things seem to work OK, if I make sure that Cura is set to "Save to File" (instead of "Save to Removable Drive") before quitting Cura.

Ghostkeeper, 2017-02-06:
As I understand it: The SD card reader essentially promises Cura that it can be accessed, but when Cura tries to access it Windows crashes the platform Cura runs on from underneath (Python).

It's a bit beyond our scope.

ianpaschal, 2018-02-21:
Since this is more than a year old and we don't expect to work on it any time in the near future, I'm closing this issue.

closed 2018-02-21 by ianpaschal