Ultimaker/Cura #1404: Manage materials: ignoring custom profiles

fbertram, 2017-02-05:
observed with beta 2.4:

I have created a custom material profile, which Cura saved in AppData/Local/cura/materials. Unfortunately, Cura keeps forgetting this profile when I quit and re-launch Cura: while I can continue to use this profile, it does no longer show up in the list in Preferences/Materials, so that I can no longer select this material and make edits to it.

My material profile defines a new brand (PolyMaker), making this issue very easily observable, as the brand disappears from the main screen's material drop-down.

This is a quite critical bug, as my material is 1.75mm and all the pre-defined materials are 2.85mm.

nallath, 2017-02-06:
How did you create a custom material? By duplicating it? 

Could you attach the files that you created?

Ghostkeeper, 2017-02-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/1104.

closed 2017-02-06 by Ghostkeeper