Ultimaker/Cura #1406: Cooling Fan controls are too limited

cinderblock, 2017-02-06:
The cooling fan options only allow **increasing** cooling over the course of the print.

> At the layers below the fan speed gradually increases from zero to regular fan speed

I'd like to start with the fan **on** for a layer and then turn it **off**.

In particular, I've found that turning on the cooling fan for layer 0 makes for much more consistent raft breakaways.

Unfortunately, the only way I've found to accomplish this currently is manually editing of the G-code.

fieldOfView, 2017-02-06:
You can use gcode postprocessing for this. See Extensions -> Post Processing -> Modify G-Code, Add a TweakAtZ step and you should be able to set fan speeds at arbitrary layer numbers

cinderblock, 2017-02-07:
Thanks for the pointer. Unfortunately there are bugs in TweakAtZ that exactly prevent what I'm trying to do. Also, I have a larger print file (~30MB) and it takes ~20 seconds for the post processor to run even though Cura only takes a couple seconds to generate the base G-code.

Regardless, I think there would be value in adding more cooling fan options. Why does it always start at 0 speed at layer 0 and go up linearly from there? Some prints might only want to start increasing fan speed after ~3rd layer. Or maybe a logarithmic curve would be better. Or start at 10% and go up from there. Or vary the fan speed based on layer size (I realize there is an option that does something close to this, based on layer time).

Alternatively, incorporating the TweakAtZ features into Cura[Engine] would be nice, imho. I'd love to be able to tweak any (or most of) of the cura settings at arbitrary heights and not just the ones TweakAtZ decided to make available. Also, running multiple TweakAtZ scripts linearly increases the post-processing time which I believe could be dramatically reduced if implemented in the CuraEngine.

Ghostkeeper, 2017-02-07:
We've had more requests for this. I think it would be a good addition.

Ghostkeeper, 2017-02-07:
I've just tested this and it seems to work already. Setting the speed to 0 is an edge case, but with these settings I can make the fan start at 100% and gradually decrease to 1%:

Fan speed: 1%
Initial Fan Speed: 100%

The result is:

    ;LAYER:0
    M106 S255
    ...
    ;LAYER:1
    M106 S170.8
    ...
    ;LAYER:2
    M106 S86.7
    ...
    ;LAYER:3
    M106 S2.5

Doesn't that work for you?

> Or maybe a logarithmic curve would be better.
We've noticed that most fans, at least the ones on the Ultimaker printers, don't respond to PWM linearly. Specifically, there is a border around 40% where it suddenly becomes a lot more effective than a few percentage points lower. Different response curves would definitely be an improvement. But its usefulness is marginal since the profiles typically use full fan or none, so it's only to make the gradual increase more gradual and to make the fan behave more as the user expects it when the user changes the fan speed.

BagelOrb, 2017-02-07:
I think that the nonlinearity in the fan response should be fixed in the firmware on the printer.

cinderblock, 2017-02-07:
@Ghostkeeper I didn't see the new `Initial Fan Speed` that was added in 2.4. That indeed does what I want (albeit with the bug that it doesn't go quite to zero but 1 is close enough to 0 that it doesn't run).

@BagelOrb There are two/three places where the nonlinearity comes into play.
1. Does the `M106` command PWM duty cuty, RPM, airflow, watts of heat removed, or something else? If a printer does one but the user expects another, that is effectively a non-linearity.
2. In most printers, `M106` seems to controls fan PWM. There are however mechanical nonlinearities: like friction or when the PWM value doesn't break the fan's stiction. I believe you're talking about these issues and I think you're expecting the `M106` to control the fan's RPM. (Some servo loop in the firmware would control PWM to maintain correct RPM). For the record, I think `M106` should control RPM, not PWM.
3. If I want a gradual change from 0 to 100% (aka 255), should it step linearly from 0 to 255? I'd argue that a logarithmic progression would be more "gradual", just like good LED dimming schemes. By "gradual" I mean "one step", at/near the beginning or end, should feel like the same % change in cooling power / airflow / RPM. This is the nonlinearity I was suggesting would be handy to have control over in Cura.

Since the original reason for this issue has been resolved in Cura 2.4, should I close this? The discussion has since moved on and there are some points to consider, imho. Maybe change the issue's title?

BagelOrb, 2017-02-08:
1. It should do airflow. That is the function of the fan. Just like we
control the heater block by temperature, not by energy or voltage.

3. It should step linearly. Light intensity is a function of the wattage
cubed I think. Similarly air flow is not a linear function of voltage or
RPM.

M106 S127 should be exactly half the air flow of M106 S254. The
'logarithmic' scale you talk about should be present in the firmware, not
in Cura.

<https://github.com/Ultimaker/Cura/issues/1406#issuecomment-278173638>, or mute
the thread

cinderblock, 2017-02-08:
1. Do you know of any printers that actually calibrate `M106` to airflow? I think this would be rather difficult to do for very little gain, if at all. Worth exploring regardless.
2. 
3. I believe I've failed to explain the logarithmic scale that I'm talking about. I agree that `S127` should be half (within rounding error) the airflow/rpm (depending on how `M106` is defined) of `S255`. I'm saying if, for instance, you started at 0 airflow on layer 0, and wanted to be at 100% by layer 4, a logarithmic progression instead of a linear one could possibly make more sense.

Linear:
 - Layer 0: `M106 S0`
 - Layer 1: `M106 S64`
 - Layer 2: `M106 S127`
 - Layer 3: `M106 S191`
 - Layer 4: `M106 S255`

Logarithmic (each layer gets ~4x more airflow/rpm than the previous)
 - Layer 0: `M106 S0`
 - Layer 1: `M106 S4` (Yes, the first step is technically not logarithmic because logarithmic scales don't go to 0. Close enough)
 - Layer 2: `M106 S16`
 - Layer 3: `M106 S64`
 - Layer 4: `M106 S255`

Now, I'm not sure if this would actually be better. It's just an option to explore.

My point about driving lights is that using a logarithmic scale for dimming/fading *looks* better to humans. I'm just guessing that the physics of the heat removal might line up with or be better suited to a logarithmic scale.

cinderblock, 2017-02-09:
@Ghostkeeper

I managed to trick CuraEngine to print `M106 S0` by setting the `Fan Speed` to `0.01` which after rounding to 1 digit after decimal point is 0.

#### Another issue, there seems to be an inconsistency in layer numbering:

If I set `Regular Fan Speed at Layer` to `1`, I'd expect the first layer of the part, `LAYER 0` in g-code, to use the `Initial Fan Speed` and gcode `LAYER 1` should be at 0 fan speed.

However, using `Regular Fan Speed at Layer` `1` jumps immediately to the final fan speed, skipping the initial speed altogether. I need to set it to `2` to get the behavior I want (the fan is on for exactly one layer).

Ghostkeeper, 2017-02-09:
> I managed to trick CuraEngine to print `M106 S0` by setting the `Fan Speed` to `0.01` which, after rounding to 1 digit after the decimal point, is `0`.

:laughing: Very good!

Regarding the layer numbering, we explicitly set that to start counting at 1 since the rest of the interface of Cura does this too (namely the layer view). Also, most non-programmers (and programmers in a non-programming context) start counting from 1 in daily life. 1 refers to the first layer. Except in the g-code, but that's an unfortunate aspect of how the g-code is defined.

The logarithmic function applied by our brain is unrelated to filament cooling to some amount of airflow. And neither are related to the original issue, so I'm closing this since the original issue was resolved by our new fan setting.

closed 2017-02-09 by Ghostkeeper

cinderblock, 2017-02-09:
I figured it was inconsistency between 0-index and 1-index. I feel like there were some other settings that used 0-index and that was my main source of confusion but I am not finding them now.