Ultimaker/Cura #1411: Installing Cura 2.4 removes machine profiles from 2.3

cinderblock, 2017-02-07:
I just updated to the 2.4 beta. Loving all the changes so far.

One nice thing was that all my 2.3 profiles and such were imported by 2.4 and I can start using it immediately.

I wanted to do a little direct comparison to the old Cura for my own sanity. Opening my still installed Cura 2.3.1, all my machine and material definitions have been removed!

Does cura share a settings folder between the versions?

nallath, 2017-02-08:
Yeah. This is an oversight from our end. We have an issue on our own tracker regarding this. I hope that we can adress this for the next release.

Ghostkeeper, 2017-02-09:
Since we have this in our tracker, I'd like to keep this issue open and link to it from the tracker. That way we won't get more duplicates regarding this.

nallath, 2017-02-09:
Oh, sorry. You're right.

ianpaschal, 2018-02-21:
Since this is more than a year old and we don't expect many people to upgrade 2.3 -> 2.4 now, I'm closing this issue.

closed 2018-02-21 by ianpaschal