Ultimaker/Cura #1413: additional speed setting for top layers

maukcc, 2017-02-08:
Is it possible to implement an additional speed setting for the 1st top layer?
It doesn`t matter whether it is a bridge layer, a top layer over infill or a top layer over support, You will want this layer to be printed slower, and preferably with more flow, but that is a different story all together :)

BagelOrb, 2017-02-08:
If you define a top layer to be one with air immediately above it then any slanted surface will have both top skin and normal skin. You would then divide those areas and print them in two passes, rather than printing the whole skin area as one.

Hopefully we can find a solution to this problem in the future, but for now this feature request is not something we will implement.

closed 2017-02-08 by BagelOrb

maukcc, 2017-02-10:
I am not sure that I understand your proposed workaround. 

To clarify my predicament I attached the following file:
[testprintBridging.stl.zip](https://github.com/Ultimaker/Cura/files/766375/testprintBridging.stl.zip)
If you slice this with 0.1mm then the following layers should go slower to fill them propperly:
- start of support interface (9 in my case)
- start of bridge (19 in my case)
- start of top layers (30 in my case)

BagelOrb, 2017-02-10:
Suppose you tilt your model by 10 degrees so that it will be slanted.

What layers should then go slower?

Perhaps thinking about that question makes you understand the problem.

maukcc, 2017-02-10:
Ahh, got it.

The quick answer: go slower on all of them, as there is no other solution at the moment.
For the general public it is better to print a couple of minutes longer, then to fail a print :)
Then if they get more comfortable with the software they can tweak.
Time to print an object should not be a major factor, and if it is for some one, it can be reduced formidably with options like the one over at #1292
So for now this remains a serious problem.

A possible solution in the long run: 
if surface area of the 1st top layer is larger then xxx square mm, then go slower.