Ultimaker/Cura #1414: Feature request: "Support Pattern -> Sheathed Zig Zag"

FlameRunner, 2017-02-08:
The problem:
Current sheathed patterns (Grid, Triangles) make the support hard to destroy and remove.
Unsheathed patterns (like Zig Zag) do not give proper support for the support interface.

Proposed solution:
Option to generate a perimeter around Zig Zag structure.

If the description isn't clear, I can add some screenshots, just let me know.

Ghostkeeper, 2017-02-09:
So basically, Lines with a perimeter around them. Did you try Grid?

FlameRunner, 2017-02-09:
Personally I would prefer Sheathed Zig Zag than Sheathed Lines - less travels/retractions.

Grid is what I'm using now and I often have hard time destroying and removing the support.

Ghostkeeper, 2017-02-10:
I've added an internal issue to have this scheduled for a release. Seems like a useful feature.

Ghostkeeper, 2017-06-05:
Our product manager just closed this and removed it from our planning for being inactive for 3 months.

FlameRunner, 2017-06-06:
So... not happening? :(

BagelOrb, 2017-06-06:
Not any time soon :S

BagelOrb, 2018-02-28:
Made a PR for this. It was quite easy.