Ultimaker/Cura #1415: Axis Homing.

Cdschlosser, 2017-02-08:
I use several different programs for slicing and both Cura 2.3 and 15 do this.  I have caught it mid print homing the X or the Y and on occasion it will try to home the Z right through the project or at least melting the nozzle into it.  The X and Y do this but it is not an issue.  I do not have this issue with Slic3r so I have to believe it is to do with Cura.  I have a crash right now and I will take a picture when I get home.

Ghostkeeper, 2017-02-09:
Rather than a picture, could you find a g-code file that does this? If it does homing in the middle of a print, it might be a g-code command where we intend the printer to do something else but the printer interprets it as homing instead.

Also, what printer do you use?

Cdschlosser, 2017-02-10:
Never Mind I think I found a post explaining it.

The issue was due to a corrupt SD-card, which was occasionally having some garbage read from it. It turns out that Marlin will try interpret a corrupt move command like G0 X1y3.54 and still read as many numbers as it can. In this example, it would be interpreted as G0 X1.

This explains my symptoms perfectly:

X and Y always moved to (approximately) their home positions, but it was always only one of them (it's quite unlikely that both moves are corrupted).
Z was not affected because Z moves are much rarer in the G-code (only on layer change) and thus it was very unlikely that a Z move would be affected.
E was not affected since a request to move E to near 0 would be prevented by Marlin's long extrusion prevention.

Cdschlosser, 2017-02-10:
Thanks for the interest in it.

closed 2017-02-10 by Cdschlosser