Ultimaker/Cura #1416: CURA 2.4 BETA crashes on start but at a diffrent location :)

taxus812, 2017-02-08:
Same symptoms as other issues I have seen. 
Windows 10 Version 2.3.1 and 15.04 installed and working. I installed 2.4 worked first launch. subsequent launches fail with "Cura stopped working".

here is cura.log 

2017-02-08 14:14:42,119 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin FileLogger
2017-02-08 14:14:42,140 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeProfileReader
2017-02-08 14:14:42,162 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin GCodeWriter
2017-02-08 14:14:42,189 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ImageReader
2017-02-08 14:14:42,220 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LayerView
2017-02-08 14:14:42,242 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LegacyProfileReader
2017-02-08 14:14:42,265 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin LocalFileOutputDevice
2017-02-08 14:14:42,291 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MachineSettingsAction
2017-02-08 14:14:42,317 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-02-08 14:14:42,329 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PerObjectSettingsTool
2017-02-08 14:14:42,356 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin PostProcessingPlugin
2017-02-08 14:14:42,388 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RemovableDriveOutputDevice
2017-02-08 14:14:42,425 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SliceInfoPlugin
2017-02-08 14:14:42,449 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SolidView
2017-02-08 14:14:42,474 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin CameraTool
2017-02-08 14:14:42,514 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin MirrorTool
2017-02-08 14:14:42,550 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin RotateTool
2017-02-08 14:14:42,582 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin ScaleTool
2017-02-08 14:14:42,609 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SelectionTool
2017-02-08 14:14:42,639 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin TranslateTool
2017-02-08 14:14:42,674 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UltimakerMachineActions
2017-02-08 14:14:42,766 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UM3NetworkPrinting
2017-02-08 14:14:42,841 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-02-08 14:14:42,854 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin UpdateChecker
2017-02-08 14:14:42,934 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin USBPrinting
2017-02-08 14:14:42,969 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade21to22
2017-02-08 14:14:42,996 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin VersionUpgrade22to24
2017-02-08 14:14:43,026 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin SimpleView
2017-02-08 14:14:43,053 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin WireframeView
2017-02-08 14:14:43,081 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin X3DReader
2017-02-08 14:14:43,122 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XmlMaterialProfile
2017-02-08 14:14:43,151 - INFO - UM.PluginRegistry.loadPlugin [100]: Loaded plugin XRayView
2017-02-08 14:14:43,164 - INFO - UM.Qt.QtApplication.__init__ [90]: Command line arguments: {'file': [], 'external-backend': False, 'disable-textures': False}
2017-02-08 14:14:43,192 - INFO - UM.VersionUpgradeManager.upgrade [100]: Looking for old configuration files to upgrade.
2017-02-08 14:14:43,218 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-02-08 14:14:43,250 - INFO - UM.Backend.Backend.startEngine [71]: Started engine process: C:\Program Files\Cura 2.4\CuraEngine.exe
2017-02-08 14:14:43,260 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.4\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.4\\resources\\definitions\\fdmprinter.def.json', '']
2017-02-08 14:14:43,319 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-02-08 14:14:43,334 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Cura_SteamEngine version DEV
2017-02-08 14:14:43,344 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] Copyright (C) 2014 David Braam
2017-02-08 14:14:43,356 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-02-08 14:14:43,366 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
2017-02-08 14:14:43,392 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-02-08 14:14:43,426 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-02-08 14:14:43,458 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] (at your option) any later version.
2017-02-08 14:14:43,491 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-02-08 14:14:43,522 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is distributed in the hope that it will be useful,
2017-02-08 14:14:43,554 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-02-08 14:14:43,583 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-02-08 14:14:43,611 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] GNU Affero General Public License for more details.
2017-02-08 14:14:43,638 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-02-08 14:14:43,652 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-02-08 14:14:43,661 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-02-08 14:14:43,994 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3; Multi-extrusion printers are not supported
2017-02-08 14:14:45,890 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to punchtec_connect_xl; Multi-extrusion printers are not supported
2017-02-08 14:14:46,716 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_offset_x
2017-02-08 14:14:46,728 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_offset_y
2017-02-08 14:14:49,094 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker3_extended; Multi-extrusion printers are not supported
2017-02-08 14:14:50,008 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
2017-02-08 14:14:51,066 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [69]: Not attaching MachineSettingsAction to ord; Multi-extrusion printers are not supported
2017-02-08 14:14:57,393 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_head_distance
2017-02-08 14:14:57,438 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_head_distance
2017-02-08 14:14:57,572 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_head_distance
2017-02-08 14:14:57,616 - WARNING - UM.Settings.DefinitionContainer.deserialize [202]: Unable to override setting machine_nozzle_head_distance
2017-02-08 14:15:13,300 - DEBUG - UM.Backend.Backend._logSocketState [171]: Socket state changed to Connected
[... 241 lines cut ...]
2017-02-08 14:34:22,923 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] 
2017-02-08 14:34:22,938 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] This program is free software: you can redistribute it and/or modify
[... cut ...]
2017-02-08 14:34:44,981 - DEBUG - UM.Qt.Bindings.Theme.load [126]: Loading theme file: C:\Program Files\Cura 2.4\resources\themes\cura\theme.json
2017-02-08 14:34:45,144 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme C:\Program Files\Cura 2.4\resources\themes\cura
2017-02-08 14:34:45,835 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [175]: Could not get metadata of container  because it was not found.
2017-02-08 14:34:46,446 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [134]: Creating additional ui components for UM3.

taxus812, 2017-02-08:
did a clean uninstall of all old version, deleted directory's and appdata reinstalled same issue same spot.

nallath, 2017-02-09:
The logs don't show anything going wrong. 

Did you ever try to install 2.3?

taxus812, 2017-02-09:
I reinstalled 2.3 and now it started failing the at the same point. (it previously worked before the 2.4 install).   With one exception I saw a very quick OpenGl message on the screen.  

So I went hunting in the event logs and found this:

Faulting application name: Cura.exe, version: 0.0.0.0, time stamp: 0x54467a5f
Faulting module name: opengl32.dll, version: 10.0.14393.0, time stamp: 0x57899b28
Exception code: 0xc000041d
Fault offset: 0x0000000000009cc2
Faulting process id: 0x2814
Faulting application start time: 0x01d282494a0769c9
Faulting application path: C:\Program Files\Cura 2.3\Cura.exe
Faulting module path: C:\WINDOWS\SYSTEM32\opengl32.dll
Report Id: 211df1f6-cdc9-4c94-9831-59eac5396507
Faulting package full name: 
Faulting package-relative application ID: 

and this one 

Fault bucket 120515647699, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: Cura.exe
P2: 0.0.0.0
P3: 54467a5f
P4: opengl32.dll
P5: 10.0.14393.0
P6: 57899b28
P7: c0000005
P8: 0000000000009cc2
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER13D6.tmp.WERInternalMetadata.xml

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_Cura.exe_3b7a4dc96517211f44da2b4054f0ef578a2c5_0535297f_24391f31

Analysis symbol: 
Rechecking for solution: 0
Report Id: c9e24c59-03ea-459a-b963-58f3731ceb6a
Report Status: 1
Hashed bucket: daa8b3b445db65f2115442604a2aba55

Speculating that the OpenGL driver was now "damaged" in some way I uninstalled and reinstalled my video driver.  The 2.4 install now loads.

Ghostkeeper, 2017-02-09:
Huh. So the end result is that the video driver got damaged somehow when Cura got opened or between the first and second time you launched Cura.

In any case, I don't think it was Cura itself that did this, as Cura doesn't have any access to admin-protected files like that driver. I'm glad you found a solution for yourself.

closed 2017-02-09 by Ghostkeeper

taxus812, 2017-02-09:
um yeah it was the install of 2.4 that broke it for both 2.3 and 2.4.  No other application was affected. Reinstalling OpenGL  simply reset whatever got damaged.