Ultimaker/Cura #1417: Upgrade to Cura 2.4 seems to break profile definitions and other issues

cinderblock, 2017-02-09:
I've recently upgraded to Cura 2.4 which seems to have imported my profiles correctly. I did also open 2.3 again and that might have caused my problems.

However, getting ready to slice and the "Save to File" button is greyed out. Turns out an error was being displayed:

> Unable to slice with the current settings. The following settings have errors: Prime Tower Y Position, Regular/Maximum Fan Speed Threshold

There are a couple things that seem wrong here:
1. I'm using a single head extruder and as I understand it, the Prime Tower is only used for dual extruders. Why is it complaining about the prime tower position and only the `Y` position at that?
2. I have not adjusted the `Regular/Maximum Fan Speed Threshold` since updating. Why is it erroring now?
3. It seems to complain about anything shorter than `15s` for `Regular/Maximum Fan Speed Threshold`. I feel like I should be allowed to set it to `0s`
4. Once I do set `Regular/Maximum Fan Speed Threshold` to an acceptable value, the error for the `Prime Tower Y Position` goes away and slicing works.

cinderblock, 2017-02-09:
Well, I'll answer some of my own questions: `Regular/Maximum Fan Speed Threshold` has a lower limit of `Minimum Layer Time`. It used to just be a warning. Now it is an error. I still don't get why the error message was referencing the priming tower.

Ghostkeeper, 2017-02-09:
Ah, I think we might still display an error message if the prime tower is in a location that collides with the disallowed areas or the border of the build volume, even if you're using single extrusion and the prime tower is hidden.

We don't have a good solution with our current architecture for the case where one setting needs to be higher than the other, such as `Support Tower Diameter` and `Support Tower Minimum Diameter`, or `Regular/Maximum Fan Speed Threshold` and `Minimum Layer Time`. Currently we limit one of the settings, in this case `Regular/Maximum Fan Speed Threshold` that it must always be higher than `Minimum Layer Time`. We don't say that `Minimum Layer Time` must be lower than `Regular/Maximum Fan Speed Threshold`. As a result, only `Regular/Maximum Fan Speed Threshold` gets coloured red and put in the notification, while really it's the combination of the two settings that is at fault.

hyOzd, 2017-05-18:
That was a really frustrating problem until I found this report. Please solve this.

EldRick, 2017-06-25:
I nfixed this issue by Exporting and then Importing the Profile. The issue went away.

closed 2017-07-27 by nallath