Ultimaker/Cura #1418: Material Profile's default temperature can't be overriden with higher temperature

Bobb1, 2017-02-09:
When slicing a model with a Material Printing Temperature set above the Material Profile's Default Temperature, the resulting gcode uses the Material Profile's Default Temperature.

Bobb1, 2017-02-12:
Actually, I don't know where the extruder Temperatures in the saved gcode are coming from.  I can't get ABS to output gcode with 245C setting.  Also, I set PLA to be extruded at 200C, and the resultant gcode is 205C.

fieldOfView, 2017-02-13:
There's a new setting in Cura 2.4 to allow you to set the nozzle temperature for the first layer. By default this is set to the normal nozzle temperature + 5.

Ghostkeeper, 2017-02-13:
If that was the problem, this is a duplicate of https://github.com/Ultimaker/Cura/issues/1315.

Bobb1, 2017-02-13:
With the comments from #1315, that does explain the behavior I've been seeing.  However, if the Setting Visibility is only turned on for Printing Temperature, I would prefer to have the other three temp settings ignored.  In other words, only act upon temperature settings that have been made visible.

Ghostkeeper, 2017-02-13:
If this is a duplicate of #1315, you should see the temperature restore to normal on the second layer.

Bobb1, 2017-02-13:
It is a duplicate of #1315.

closed 2017-02-13 by Bobb1