Ultimaker/Cura #1420: Custom Printer Printhead Settings ignored

StefanKi, 2017-02-10:
I have a custom printer.

The 0,0 point is at the front left.
The maximum (200,220) is at the back right.
There is no print bed in the first 20mm in Y.
So I can only print 0-200 in X and 20-220 in Y.
[image]

My Machine Settings are following:

Printer Settings:
X (Width): 200
Y (Depth): 200

Printhead Settings:
X min: 0
Y min: 20
X max: 200
Y max 220

If I put a part at the front left, the print starts at 0,0. But there is no build plate. The build plate starts at 0,20.
If I put a part at the back right, the print don't fits the back, but is 20mm apart from the back.

I think this is a bug - or did I make a mistake?

nallath, 2017-02-10:
The um2go has a smilar issue. In that case we solved this by changing it in the firmware. The home position is set to be (0,-20), so (0,0) is actually the first part of the plate where a print can be done.

Ghostkeeper, 2017-02-10:
You could also solve this by adding a disallowed area. This is only possible with an actual machine definition file though (`.def.json`). Not with the "Custom FDM Printer" option in the Cura GUI.

The disallowed area is rendered as a shadow in Cura. It prevents people from placing items on it.

StefanKi, 2017-02-10:
So my interpretation and solution of the custom printer configuration is correct?

Both "workarounds" aren't very satisfying. I think I will try both versions to find out which one is best for me. (Of course not simultaneously...)

closed 2017-02-28 by StefanKi