Ultimaker/Cura #1423: ungroup is not working (disabled)

bond4u, 2017-02-12:
I'm trying to print "pump hand drill" from tv and "ungroup" option is greyed out in rclick menu.
Not sure why.
[image]

fieldOfView, 2017-02-12:
Did you load these parts as a single file?

Ungrouping only works for models that were grouped inside Cura. It does not break apart submeshes from a single model.

peteruithoven, 2017-02-13:
So to clarify, this issue is on reimplementing the take apart functionality?

Ghostkeeper, 2017-02-13:
The "Split object into parts" function from legacy Cura did this, but it was never ported to Cura 2.

We have this feature on our tracker. Currently planned at 2.6 (but I can give no guarantees on people moving that around).

By the look of the shadows underneath that model, it is indeed just one model file and so Cura is not able to ungroup it. But let's wait for Bond's answer to confirm this first.

bond4u, 2017-02-13:
Yeah, these objects are from single file (http://www.thingiverse.com/thing:731086)
It says "ungroup models", not "ungroup files" ;) so I assumed it can "split" them as old cura did.

Ghostkeeper, 2017-11-20:
Let's mark this as a duplicate of https://github.com/Ultimaker/Cura/issues/2052

closed 2017-11-20 by Ghostkeeper