Ultimaker/Cura #1425: Scale issue. Skirt didn't follow model size

Wisnaeme, 2017-02-12:
I've had this happen a few times now.
I scaled up a model and the skirt stayed the same size so the model was printed over the smaller skirt.

Ghostkeeper, 2017-02-13:
I've never seen this. It is also very strange from a technical viewpoint, since the scaling is applied by the front-end and then the scaled model is sent to the back-end. The back-end computes the skirt and the main g-code, without knowing about the original scale. So if the scale of the skirt is different from the rest, then the back-end would've gotten two meshes and somehow split the g-code apart or something.

So what I think is more likely is that the model is not entirely flat on the bottom. It draws the skirt only around the bottom layer, so if the bottom is not flat it will draw the skirt only around the parts that protrude furthest downwards.

Could you attach a model with which this happens? Or perhaps a screenshot of the first layer? Or both?

BagelOrb, 2017-03-08:
It could also be the case that you didn't eject your USB before pulling it out of your PC.

Closing due to inactivity.

closed 2017-03-08 by BagelOrb