Ultimaker/Cura #1427: One at a time not working

MarcusWolschon, 2017-02-13:
Cura 2.3.1
Printer profile: Ultimaker II extended - modified for 1.0mm nozzle size
printing profile and a simple test file attached.
[one_at_a_time.zip](https://github.com/Ultimaker/Cura/files/772091/one_at_a_time.zip)

Even though the 4x3 models are 33mm apart, they are sill printed at the same time.
The "one at a time" setting is completelly ignored.

Ghostkeeper, 2017-02-14:
Did you perchance happen to *group* these models?

It is an "undocumented feature" that groups of meshes are printed at the same time. It's a relic of how it was communicated between front-end and engine, but it allows for more precise control over what gets printed in what order.

fieldOfView, 2017-02-14:
"One at a time" only applies to separate models. If you have multiple objects in one stl file (as you have in your attached file), Cura sees that as a single model (and always prints it all at once).

MarcusWolschon, 2017-02-14:
This limitation (multiple bodies/shells loaded from the same file) should be made clear to the user (e.g. in a tooltip or other help text for the "one at a time" function).
It is not expected behavior to the uninitiated user.

Ghostkeeper, 2017-02-16:
It is also the case for scaling/rotating/mirroring and per-object settings.

Ghostkeeper, 2017-02-16:
The 3D format that you're using doesn't define multiple shells and such. It simply lists a number of faces. Whether they belong to the same shell is up for interpretation.

There are formats that do define multiple shells. 3MF is one of them. Cura supports that and loads them as separate meshes. Printing one at a time would then also print those one at a time.

closed 2017-11-13 by ChrisTerBeke