Ultimaker/Cura #1428: Taz 5 build envelope is off in view.

partsoven, 2017-02-14:
Setup the Taz5 in this new Cura and the noticed build window and the machine are way off from each other.

Ghostkeeper, 2017-02-14:
We don't have a Taz5 definition in Cura, as far as I know.

If the build plate mesh visualisation is off, you can either move the coordinates in the actual STL file, or modify the "platform_offset" metadata property like we do here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker2.def.json#L16 . It is an offset applied to the build plate (X, Y, Z coordinates).

Ghostkeeper, 2017-02-14:
Perhaps you're using Lulzbot's edition of Cura?
They have a repository over here where you can ask their developers: https://code.alephobjects.com/project/profile/10/
In the meanwhile, I'll page @kakaroto ... Perhaps he can help out.

kakaroto, 2017-02-21:
Yeah, the user was probably using a development build of the lulzbot's edition of Cura2.
Considering that our Cura2 is still in alpha and not yet ready for use, I would close this issue. 
Once we fixed most of our issues and it has gone through proper testing, we will release a beta and will be ready to accept bug reports on it.

nallath, 2017-02-22:
Will you also merge your definitions into "our" Cura?

kakaroto, 2017-02-22:
That's an interesting/hard question @nallath which I don't really have an answer for right now. We do plan on being as close as possible to upstream and we've prepared a bunch of pull requests (https://code.alephobjects.com/T689) which I'm supposed to send for review/merge (as soon as I stop being swamped by urgent issues).
There are lulzbot specific commits though that we don't know if you would want to have upstreamed (such as 'wipe temperature', 'probe temperature', "part removal temperature" settings for materials, the lulzbot theme, a potential toolhead selector system for the printers, the build script which would grab the hundreds of materials/qualities files from a separate repo, etc..)
If you don't add all of our features/materials, then a user using the Ultimaker Cura2 and selecting a Lulzbot printer, would have a subpar experience, but if you are willing to accept all of our lulzbot-specific changes, then that would be great.
Right now, we have a bunch of bugfixes/features that are beneficial to all, so we will send the PRs for those, as for the lulzbot-specific stuff, as it's still incomplete, it wouldn't be ready to be upstreamed right now anyways, but once that's stabilized, then I guess we can consider the question and have a discussion about it.

Anyways, that's my opinion, and I'm just a lowly consultant, what do I even know ? :p

partsoven, 2017-02-22:
I for one have 3 Lulzbot printers and 2 brand new Ultimaker 3s in my office. One Cura to rule them all would be so EXCELLENT.

nallath, 2017-02-23:
Any change that benefits our printers (or a large subset of other printers) is something that we want. That being said, the system supports enabling certain settings for certain printers (and disabling it for others). I see no issue in having settings that are only there for certain printers. 

The theme is of course not something that we should merge, unless we have a theme selection method.

awhiemstra, 2017-02-23:
I think having everything in mainline that is needed to get the Lulzbot printers printing correctly would be nice.

Ghostkeeper, 2017-02-23:
I agree with nallath and awhiemstra that all of these changes are desirable in Ultimaker's Cura as well and we'd gladly accept your pull requests. I see two (minor) obstacles:

* We'd need a theme selection setting before the theme can be accepted. And since Lulzbot probably wants its own theme to be selected by default and Ultimaker wants the other, we'd need some way to set the default for that preference in the build process, so that both teams can build their applications from the same base (though of course they'd still be separate forks).
* If we set up the build system to pull in profiles automatically from a separate repository of Alephobjects, then we'd probably want to pull in a static commit hash so that your changes can't possibly break Ultimaker's build. Whenever we make a new release (or whenever it is desired to have updated profiles) we'll move the commit hash to a more modern commit that is tested to work. This is what we do with other dependencies as well. The "extra repositories" CMake parameter in cura-build already sort of provides this functionality.

kakaroto, 2017-02-24:
That's great to hear, if you are open to that, we'd love it. I really like the idea of having a single source that is merged in both directions and eventually releases being pretty much 100% identical.

As for the theme, someone had a great idea in today's LuzlBot meeting, if the theme can be changed dynamically, then the theme could change depending on the machine that you select. So when you select a lulzbot machine as the currently active one, it would switch to the lulzbot theme, otherwise, go back to default theme. This would get rid of the requirement for a theme selector (which would make things harder if the users select a theme and it gets changed from under them when they select a different printer).

Note by the way, that we already have a theme selector option in the Cura General configuration (https://code.alephobjects.com/rCTbe8d6e8f41ee9f1033d1ae29ca4add01d0547347), it's one of the commits that I need to send a pull request for (I wanted to first separate the lulzbot theme from the addition of the theme selector option, so we could send the commit in a PR without the lulzbot theme, but if you'd accept both, then that makes things easier).

nallath, 2017-02-27:
I really want a "dark" cura theme, so having multiple themes is not really an issue for me.

Ghostkeeper, 2017-03-01:
With regard to other applications depending on the Uranium framework and the framework's original intent as 'basis' for Ultimaker's desktop GUI applications, it's probably not good to make the theme switchable only by machine selection. Having a manual UI element in the preference would also allow stuff like Nallath's dark theme.

closed 2017-07-27 by nallath