Ultimaker/Cura #1431: Enhancement Req.: New Print Sequence "1/10" (2.5mm) at a time"

ThomasGreska, 2017-02-15:
Hello, 
when printing a set of chess figures I found, that it would be useful if the selection "Print sequence: All at once | One at a time" would be completed by a "1/10" (2.5mm) at a time".
With "All at once" the hotend spends a lot of time travelling around, with "One at a time" you need to leave huge gaps between the objects to avoid collisions.
This slices of 1/10" of an inch you could go around the field without beeing to much aware of colissions but you can print 15+ layers at a higher speed.

Btw: Great software!

Ghostkeeper, 2017-03-06:
We had a community developer 1.5 years ago that made something like this, but it was a very difficult change and his implementation only worked in very specific cases. In the end it was not implemented.

Patola, 2017-03-06:
@Ghostkeeper do you have the reference, e.g. the PR or something? Maybe the code can be worked upon.

Ghostkeeper, 2017-03-07:
The repository is here: https://github.com/sjoerdtimmer/CuraEngine/tree/feature_towering

Ghostkeeper, 2017-06-05:
Our product manager just removed this issue from our planning for being inactive for 3 months.

Ghostkeeper, 2019-12-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/6642 now.

closed 2019-12-03 by Ghostkeeper