Ultimaker/Cura #1433: QT 5.6 and python 3.5 bump

kakaroto, 2017-02-16:
Hi,
There was recently a bump to QT 5.6 and python 3.5, all I could find are these 2 commits : https://github.com/Ultimaker/Uranium/commit/53570e0aa0590578ec17f28a60335cd05c208223 and ba0fba08444746a6952508d58bc8c03312eeb48d but they don't explain why it had to happen, and they only seem to reference this "CURA-2150 " issue which is not a github issue.
Can you explain why this bump had to happen ? Which functions/APIs do you use/depend on which are only available in QT 5.6 and python 3.5 ? 
This is causing us various issues and we want to keep Cura's dependency on 3.4 but if we don't know what would be affected if we release Cura with 3.4 as dependency, then it's just going to be very hard to track down the changes that need to be done to backport them to 3.4, etc..
Is there any way to get more information on this "CURA-2150" issue and details (commits) on which changes made cura have a hard minimum dependency on python 3.5  and qt 5.6  ?
Thanks.

nallath, 2017-02-17:
The issue describes the following;

_Currently Cura requires PyQt 5.5 plus an important patch which fixes nasty deadlock problem. This custom version of 5.5 makes building and running Cura a huge pain because it requires that PyQt 5.5 be patched and recompiled instead of just relying on the packaged versions of PyQt from Riverbank Computing, your Linux distro or from some other common source online.

Instead of wasting time using 5.5+patch we should just go to 5.7. Our devs on Linux have already been using 5.7 for some time now without major problems. The more interesting platform here is Windows. To reduce risk, we can locally make a test Windows 64bit installer build and let QA try it out for a day or two.

Implementing this issue should be a matter of bumping the PyQt version number in cura-build and also installing 5.7 in each build environment using a standard PyQt package/wheel/installer.

Completing this issue will also fix the important issue CURA-2045 "32 bit: Swapping between Preferences tabs cause Cura to crash"._

We didn't upgrade this lightly, but  as you can read from the issue, it was something that had to be done.

fieldOfView, 2017-02-17:
That explains the bump to PyQt 5.7, but not the bump to Python 3.5.

AFAIK the switch to Python 3.5 is being made to support type hinting, which will help to prevent bugs. See the python_type_hinting branches:
https://github.com/Ultimaker/Cura/tree/python_type_hinting
https://github.com/Ultimaker/Uranium/tree/python_type_hinting

There's also the fact that Canonical switched to support only Python 3.5 since Ubuntu 16.x (and the distros based on it).

sedwards2009, 2017-02-17:
The type hinting stuff was merged to master this week, BTW.

Another couple of reasons as to why both were upgraded at the same time:

* (binary) builds and packages for new versions of things like PyQt and numpy etc tend to only support the latest and maybe only one previous Python version. i.e. Python version support is limited. This encourages you to keep your major dependencies fresh or accept to keep them all roughly the same age. In other words, using a recent PyQt for example, with an old Python is a PITA, especially on Windows.
* Upgrading a major dependency can introduce regression risk which requires regression testing. It makes sense to upgrade 2 dependencies and then test once, than to upgrade separately and test two times.

kakaroto, 2017-02-21:
Ok, thanks everyone for the responses. That answers my question so I will close the issue.
The issue I see with upgrading python is the problem of dependency on systems which do not have it yet (debian for example). We are already bundling Qt and PyQt and the other dependencies but python isn't bundled in the deb, and doing so might make it even more complicated to build. And we are already thinking of changing the build system in order to decouple qt/pyqt from the deb, otherwise the deb would never be approved in debian as it is against their best practices.
I understand the need for bumping PyQT, thanks for answering that (I do have that specific constant-ui-freezing issue on my system). The need for bumping python is not immediately clear to me though. I understand type hinting is nice to have, but it's not required (and type hinting syntax isn't incompatible with 3.4), I understand also the reasons outlined by @sedwards2009 and those are more convincing for doing the switch.
I have the reflex of trying to be as backward compatible as possible so users can make my software work without being bleeding-edge (and I've had headaches in the past because of that), so I was looking for some major reason (like for PyQT) for the switch to python 3.5. I guess we'll be working on integrating python in the cura build script. Will send pull requests for that soon I guess.
Thanks!

closed 2017-02-21 by kakaroto

sedwards2009, 2017-02-21:
> and type hinting syntax isn't incompatible with 3.4

Actually, I'm pretty sure it is compatible. We're still using the syntax that has been around for years. It is Python 3.6 which introduces some new syntax (types for vars). It may be possible to install the `typing` module via pip for 3.4.

...just an idea.

kakaroto, 2017-02-21:
Yeah, I know it's compatible, I said "isn't incompatible" :) because I meant it as "isn't problematic".

sedwards2009, 2017-02-21:
ok. I didn't read the double negative close enough. :)