Ultimaker/Cura #1434: Maker Select 3D/Cura 3.21. "The printer is not connected"

TTVert, 2017-02-17:
win10 64Bit, selected a Prusa i3 but also tried to create a custom printer to no avail, Works fine on Cura 15.0.4.6 and I confirmed I can communicate w/ COM3 at 115200. It seems others have this same issue and I'm not sure where to look. Looking at the log files in USBPrinting don't even mention it looking to any port. It appears this is all auto configured now but I cannot find a log indicating what it is trying to do or where it is failing.

I've attached the two log files I'm aware of that may have something to do with this but any other help would be appreciated.

Thanks
Dave

nallath, 2017-02-17:
Could you attach the logs?

Ghostkeeper, 2017-02-17:
It should be saying something like:

    2017-02-17 12:29:51,659 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [308]: Attempting to connect to /dev/ttyACM0
    2017-02-17 12:29:52,277 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [322]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
    2017-02-17 12:29:58,798 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [322]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 250000
    2017-02-17 12:30:00,311 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [348]: Correct response for auto-baudrate detection received.
    2017-02-17 12:30:00,321 - INFO - USBPrinting.USBPrinterOutputDevice._connect [355]: Established printer connection on port /dev/ttyACM0

What version of Cura are you using? We haven't released any version 3.21.

TTVert, 2017-02-17:
Sorry is the skirt the dotted line or the purple?  I believe either would be fine.  I think i understand better now after some more reading. It seems you may be able to set a global printer config that will print a skirt around any think you print.  With this particular printer it homes X/Y and then prints a glob, retracts and drags that bit of filament to prevent curling and clean up any residue.  Is this though process correct?
Thanks
Dave

TTVert, 2017-02-17:
My mistake guys.  I had to zip them as they didn't support uploading the logs directly.  I attached them to the first post.  I certainly didn't see any USB printing lines in the logs when i looked.

fieldOfView, 2017-02-17:
Those are ... not the logs.

You can find the logs by going to Help -> Show Configuration Folder... . An explorer window should open, and the folder will contain a file named "stderr.log". That's the one we're interested in.

TTVert, 2017-02-17:
Yeah Sorry I fixed it, hoped to fix it before anyone replied LOL..

TTVert, 2017-02-20:
Did you see the logs?

Ghostkeeper, 2017-02-22:
The log doesn't mention any attempt to connect via USB at all, like you said. Ctrl+F on "USB" yields 9 results, all of which are about the plug-in being loaded at program launch.

One thing that you can try is to remove other Arduino-like devices from your computer (other than your 3D printer). Then restart Cura. It sometimes gets confused by other devices that speak the Arduino language.

Also, during the installation of Cura, did you select to install the Arduino drivers in the install wizard? It's on by default (so the drivers get installed) but if you turned it off that would explain why it can't make a connection.

TTVert, 2017-02-22:
I have nothing else attached to any port aside from a USB keyboard/mouse.  This pc has no other COM ports (Internal or external) either.  Yes it did install the arduino driver.  I assume that driver controls the internal serial port in my printer (Must be an arduino board?)   I've saved my machine config which is the only thing I'd hate to lose so I'll give it a go and report back.  You'd think even w/o a driver installed it'd still try to talk to the COM port and maybe just have overrun or other communication issues.  It's not even attempting to connect. 

Thanks for the help
Dave

Ghostkeeper, 2017-02-23:
It's correct that the device driver installation doesn't pop up the second time (I think). The Arduino driver software hasn't been updated for Windows for quite some time.

Ghostkeeper, 2017-02-23:
There is very little we can do about this since we don't have this Maker Select 3D, nor do we release a Cura v3.21. I hope that someone with this printer can verify this issue and propose a fix.

TTVert, 2017-02-23:
The thing is I've uninstalled 15.04.6 and reinstalled twice, each time it has reinstalled the driver.  2.x installs haven't once attempted or requested to reinstall the driver.  Not sure it matters but it's like 2.x isn't even looking for any device as the logs show so its not like it matters what device I'm using.  It's just trying to talk to a COM port as far as I understand so the actual device shouldn't matter much should it?  I made a mistake it was 2.3 and 2.4 I tried.  No idea where I got 3.2.1 from.

awhiemstra, 2017-02-27:
Probably you mistyped "2.3.1". In any case, the reason you're not seeing a dialog for 2.x installs is that we use the silent install option of the installer. It should still install the driver though.

TTVert, 2017-02-27:
YEah I did mistype.  I fixed the title but it ins't reflecting this.  Even if it is installing a driver there is no mention of it even trying to call the device in the logs.

ntoff, 2017-03-01:
Same issue here on windows 10 in 2.4, it doesn't even seem to be trying to open a connection to anything. Repetier host works, simplify3d works, pronterface can connect, cura 15 can connect but only when the com port is specified.

2.4 on Windows 8.1 will connect whenever it feels like it which is generally never when I want it to (like right after I've closed the connection in repetier host) and never when I actually want it to. Almost like it's the scared little kid at school always looking to the cool kids to know whose jokes to laugh at and where the good spots are to hang out, i'll only connect to something after the cool kids have connected to it.

TTVert, 2017-03-02:
I think we're just stuck on an old version.  Regardless of the printer model I assume this would the same result given it doesn't even try to look for anything on any port.

ntoff, 2017-03-02:
yeah  I'm not using a maker select, I'm using a couple of mendel90's. It was better when they gave you the option to which serial port to connect to back in cura 15, at least that worked.

TTVert, 2017-03-02:
Agreed. I'm going to see what this does in different OSes and see if I at least can see in the logs if the problem follows the OS or not.

ntoff, 2017-03-02:
There's discussion here: https://www.reddit.com/r/3Dprinting/comments/5g76cs/confused_about_usb_printing_in_cura_231/ on making it list other devices rather than only usb ones but even that doesn't work for me now so I don't know what's changed from 2.3.1 to 2.4 if anything (may be one of the many auto silent updates windows 10 does to itself)

TTVert, 2017-03-02:
I'll read up on this.  I've not gotten any 2.x versions to work. 
Dave

TTVert, 2017-03-02:
I did make some progress.  I changed the line   def getSerialPortList(self, only_list_usb = False): to True and checked the registry key it is referencing.  The COM port is there and it listed as \Device\VCP0.  The fact that is says it is not arduino based tells me it is talking.  Not sure why it says it is not arduino based however.  Wonder if I need to use a different driver.  Right now it's using an FTDI supplied driver (v. 2.12.24.0) , I believe it is the driver installed via cura 15.x but I could be wrong.  

It now is trying to communicate as it resets my printer when Cura loads.  

Pertinent log info:
`2017-03-02 01:48:43,527 - INFO - USBPrinting.USBPrinterOutputDevice._connect [305]: Could not establish connection on COM3: 'Timeout'. Device is not arduino based.
2017-03-02 01:48:43,529 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [318]: Could not open port COM1
2017-03-02 01:48:43,533 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to printer with serial COM3 on baud rate 115200`

Ghostkeeper, 2017-03-02:
The time-out occurs when we get a package with length 0 or lower from the serial port. That message is then interpreted as a time-out on the serial port which is then interpreted as the device not being something we can talk with.

ntoff, 2017-03-02:
>I did make some progress. I changed the line def getSerialPortList(self, only_list_usb = False): to True and checked the registry key it is referencing. The COM port is there and it listed as \Device\VCP0.

Mine was in there and it was called \Device\Serial2 and not even that only_list_usb change would make it see it, it still didn't even _try_ to open communications. My board was a CH340 based usb-serial chip.

legoabram, 2017-03-30:
Assuming we are talking about the same issue, turns out this is a really silly bug. Under USBPrinterOutputDevice in _connect, Cura first tries to detect an arduino device using this avr_isp library called `stk500v2`. However, if this method fails, Cura never closes the port that this library uses. Every attempt to access the port at the different baud rates later in the function results in a "Access is denied" because Cura is stilling holding the port. I solved my problem by changing the following lines in USBPrinterOutputDevice.py:
```
programmer = stk500v2.Stk500v2()
try:
    programmer.connect(self._serial_port) # Connect with the serial, if this succeeds, it's an arduino based usb device.
    self._serial = programmer.leaveISP()
except ispBase.IspError as e:
    Logger.log("i", "Could not establish connection on %s: %s. Device is not arduino based." %(self._serial_port,str(e)))
except Exception as e:
    Logger.log("i", "Could not establish connection on %s, unknown reasons.  Device is not arduino based." % self._serial_port)
```
to:
```
programmer = stk500v2.Stk500v2()
try:
    programmer.connect(self._serial_port) # Connect with the serial, if this succeeds, it's an arduino based usb device.
    self._serial = programmer.leaveISP()
except ispBase.IspError as e:
    programmer.close()
    Logger.log("i", "Could not establish connection on %s: %s. Device is not arduino based." %(self._serial_port,str(e)))
except Exception as e:
    programmer.close()
    Logger.log("i", "Could not establish connection on %s, unknown reasons.  Device is not arduino based." % self._serial_port)
```
It'll be up to the Ultimaker team to finalize and bulletproof the fix, but this is what the problem is.

nallath, 2017-03-30:
We probably won't buletproof the fix, as hardly any of our printers <1% still use USB printing. 

I'm going to put your changes in there, as it makes total sense. Don't really understand why we didn't do it already.

awhiemstra, 2017-04-03:
> Don't really understand why we didn't do it already.

Probably because one expects that if a connect() method throws an exception, it would either not even be open or automatically be closed. I certainly would not expect that it keeps the socket open.

Whytehorse, 2017-06-09:
This bug affects Ubuntu 16.04 with marlin rcbugfix-1.1.x, but doesn't seem to be a problem with the marlin release

ianpaschal, 2018-02-26:
We will no longer work on or support version 2.x of Cura so this is a `Won't Fix`.

closed 2018-02-26 by ianpaschal

TTVert, 2018-02-26:
LOL thanks for the timely response..

ianpaschal, 2018-02-26:
No problemo. I've only worked at Ultimaker for 2 months and have been trying to hack through the mountain of issues here.

rflulling, 2018-03-07:
So this issue or one similar to the original post has crept up on a very real version 3.2.1 (which was stated above, doesn't exist). So now it does exist and is having issues, maintaining connection after a print. I replied to a post that was started by a Mac OS user. But other than OS his issue seemed to be the same and current. I left a full report there. Perhaps I should copy that here? Still new to the Git Hub thing, not a dev, just a user.  [64 bit Win 7 pro, M3D Micro Plus, and Power Spec i3 Mini. ]

ianpaschal, 2018-03-07:
Please make a new topic for a new issue. If you like, you can write # + the issue number in the description to reference this one.