Ultimaker/Cura #1435: Layer view legend not shown in compatibility mode

smartavionics, 2017-02-17:
As my Linux laptop doesn't support OpenGL 4.1, the layer display is working in compatibility mode. Before this new layer code was merged, a legend was being displayed that identified what was what (infill, skin, interface, etc.) That has now gone away as you can see in the following image.

Also, note that the default size of the preferences dialog is not tall enough to show all of the preferences. The dialog has to be resized to see the last preference.

[image]

smartavionics, 2017-02-17:
This is probably completely unrelated but I get this error message in the log:

file:///usr/local/lib/cura/plugins/LayerView/LayerView.qml:191: TypeError: Property 'getLayerViewType' of object LayerViewProxy(0x254da20) is not a function

Now, I know next to nothing about python but something doesn't look right with that function:

    @pyqtProperty(bool)
    def getLayerViewType(self):
        active_view = self._controller.getActiveView()
        if type(active_view) == LayerView.LayerView.LayerView:
            return active_view.getLayerViewType()
        return 0

Is the pyqtProperty(bool) correct? I would have thought that it should be int not bool?

fieldOfView, 2017-02-17:
Nice catch. The name ```getLayerViewType()``` suggests it is a method (slot), but it is declared as a property.
It should actually be ```@pyqtSlot(result = int)```. That, or LayerView.qml should use ```getLayerViewType``` instead of ```getLayerViewType()```.

Personally I would say property names should not start with 'get', so I would prefer the former fix, but unfortunately there are quite a couple of properties starting with 'get' (eg in CuraApplication). I'll prepare a PR.

fieldOfView, 2017-02-17:
See https://github.com/Ultimaker/Cura/pull/1436

Also see https://github.com/Ultimaker/Cura/pull/1437 which fixes the preferences window issue.

smartavionics, 2017-02-17:
Thanks for the quick fixes. I haven't tried these yet, does the legend now appear in compatibility mode?

fieldOfView, 2017-02-17:
No, it does not (just tested it). That will need another fix.

fieldOfView, 2017-02-17:
See https://github.com/Ultimaker/Cura/pull/1438

nallath, 2017-02-17:
All of it is fixed & merged!

closed 2017-02-17 by nallath