Ultimaker/Cura #1439: Dark Gray Border Making Print Platform Smaller (width/depth)

filipgoc, 2017-02-17:
This is probably super obvious, but

**How is the darker border on all sides of the build plate determined? What is it?**

In real terms, even though my machine size (width/ depth) is 170x160, the biggest object I can fit on the bed to slice is 160x140. Beyond that, I get an error. 

This also seems to change with machine variants, even though I put nothing in the variant other than “nozzle size”

Where does this limit come from? Any ideas? 
Thanks!

PS: the following is from the machine def.json file, and that’s the only size I ever define.

```
…
"machine_width": { "default_value": 170 },
"machine_depth": { "default_value": 160 },
…
```
[image]

fieldOfView, 2017-02-17:
The dark border shows the area reserved for build plate adhesion such as brim, raft & skirt. These increase the footprint of your models, so you cannot print as wide as your buildplate. If you want to print at the full size of your builtplate, you would have to print eg with a 0mm wide brim.

Cura 2.4 will add a setting to not print a brim, raft or skirt, making it easier to print at the full size.

filipgoc, 2017-02-17:
You're Absolutely right. 

For some reason, the 0.6mm variant has adhesion ON by default, even though it's empty... 
There probably is some adhesion cura default deep deep down (fdmprinter maybe). 

Thanks!

PS: I'm on Cura 2.4 Beta.

closed 2017-02-17 by filipgoc

Ghostkeeper, 2017-02-22:
The grey border (and build volume) is dependent on Adhesion Type, Skirt Gap, Skirt Line Count, Skirt/Brim Line Width, Brim Width, Brim Line Count, Raft Margin, Draft Shield Enabled, Draft Shield Distance, Raft Base Thickness, Raft Interface Thickness, Raft Surface Layers, Raft Surface Thickness, Raft Airgap, Extruder Prime Position X (disabled in GUI), Extruder Prime Position Y (disabled in GUI), Extruder Prime Position Z, Prime Tower Enabled, Prime Tower Size, Prime Tower Position X, Prime Tower Position Y, Ooze Shield Enabled, Ooze Shield Distance, Infill Wipe Distance, Travel Avoid Distance, Support Offset, Support Enabled, Avoid Other Parts, Support Interface Enabled, Support Extruder, Initial Layer Support Extruder, Support Interface Extruder and Platform Adhesion Extruder.