Ultimaker/Cura #1440: "um" namespace question

filipgoc, 2017-02-17:
In the XmlMaterialProfile.py, there is 

```
# Map of recognised namespaces with a proper prefix.
    __namespaces = {
        "um": "http://www.ultimaker.com/material"
   }
```

I noticed that if a material (even the ‘generic’ ones) only loads correctly if it contains xmlns="http://www.ultimaker.com/material”.

What does this do? And Why? It seems later this is use for serialization of profiles..
(Pardon my ignorance.)

nallath, 2017-02-20:
It's a namespace. You can use it to have extentions on a specification. 

See for more info: https://www.w3schools.com/xml/xml_namespaces.asp

filipgoc, 2017-02-20:
Thank you for kindly supporting the XML noob in me @nallath !

closed 2017-02-20 by filipgoc