Ultimaker/Cura #1441: How to override generic material <diameter> ? (and more)

filipgoc, 2017-02-17:
[image]

Thanks to <diameter>2.85</diameter> close to the root of generic xml materials, the generic materials list 2.85 diameter even for machines that override it to 1.75 - furthermore it cannot be edited by user. 

[image]

Maybe, <diameter>___</diameter> should not be a general property at all… still, how to override this? Or is it a bug to be squashed or a feature? 

The diameter is correctly overridden under print settings, just not under management. 
[image]

( I do wish to use general presets, I think it makes sense and corresponds to Cura’s current architecture.)

The default ```Setting (<setting key="print temperature">200</setting>)```
can be overridden when another one is nested under <machine>, which works great, but not the <diameter>… 

####  Related question
I would also like to override the <color_code> for different machines. Not color name, just the #- - -
I have a feeling it’s the same question.

ntoff, 2017-02-18:
I just create my own material profiles using this as a template

http://pastebin.com/MLSgU2Zr

I edit that file manually using notepad++ because when I add them through Cura, they don't seem to show up after restarting cura, editing them manually works fine though.

filipgoc, 2017-02-20:
Thanks @ntoff. Nice template!

However, my question is about overriding the <diameter> in the GLOBAL GENERIC profiles. 

(I have no issues making my own profiles, linking them with a specific machine, and with specific quality presets.)

These are the ones that have "generic_...." as their filename. These are a bit special. 
(example, generic_abs.xml.fdm_material)

They are the default fallback for fdmprinter, and most easily port to load generid quality profiles even when using custom color material profiles for one...

nallath, 2017-02-20:
I guess it should not really come as a suprise that we never actually tested the diameter property being used correctly (as all of our machines are 2.85). 

I'm also not that convinced that the diameter should be in the material in the first place. It's the machine that defines what the diameter is.

fieldOfView, 2017-02-20:
It makes sense to have the diameter in the material profile if you see it as the description of a SKU of a material product. If a brand would make a material in both a 3 and a 1.75 flavor, it would stand to reason that they would be different SKUs, and so have different GUIDs and different xml profiles.

Materials should specify their diameters, and machines should specify which diameters they support (with min/max?). Cura should then filter the materials to show for each printer. Ultimaker printers should not show materials that specify they are 1.75 mm, and printers which use 1.75 mm materials should not show the 3mm Ultimaker materials.

filipgoc, 2017-02-20:
> I'm also not that convinced that the diameter should be in the material in the first place. It's the machine that defines what the diameter is.

Agreed. I'm not sure material needs a diameter (even though it's been a common things across slicers for some time).

filipgoc, 2017-02-20:
For the 'generic' profiles however, I am with @nallath. Generic profiles should not have diameter information at all. They are generic after all. 

(Which gets to the original point, which was that even though the slicing settings are correctly set to 1.75, the generic material under management still shows 2.85) 

> I'm also not that convinced that the diameter should be in the material in the first place. It's the machine that defines what the diameter is.

nallath, 2017-02-20:
Well, in our case the "generic" is a bit... different. By Generic_PLA we mean "This is PLA without a specific color or attribute (such as bronze fill)". It will probably work for most configurations. 

As for special profiles, most of the settings that need to be different for a different diameter are in the qualities, which depend on machine (and on a _type_ of material). So in that case, the material doesn't need to define it.

But changing all of it now is going to take a ton of work, so as much as it pains me to say it, I don't think we're going to get arround changing that any time soon.

filipgoc, 2017-02-20:
For starters, could we simply remove the '<diameter>' line from the not many generic profiles? Any machine defines it on its own anyway. 

That does not sound like a lot of work (I'm probably missing something.)

filipgoc, 2017-02-20:
But yes, that's mostly cosmetic. Functionally, it does nothing at all there. (Potentially another reason to remove it.)

Ghostkeeper, 2017-02-22:
I'm not sure how the UM3 firmware will react when we do that...

ntoff, 2017-02-22:
I don't know though, to me the "generic" materials were always just a sort of starting platform for your own materials rather than a material to actually be used, hence my "template" thing. 

They aren't editable, only copies of them can be edited, so I figured they're more like those resume templates or angry letters to the manager templates you get with various office programs, generic starting points meant for editing and you save a new copy of it after customising it and the default settings are always there in read only mode in case you really stuff up and need a good starting point.

filipgoc, 2017-02-26:
@ntoff 

I may have the concept wrong, so please DO correct me as desired: 

From what I observed, the `generic` materials are not only templates, but also fallbacks as a base machine profile for a specific kind of material (PLA, PETG... etc). 

Example for PETG
- I have a custom machine that is listed in the generic petg xml
- i also have custom quality profiles for PETG
- these profiles will now load for any generic petg as a default for the user to play with. I can also provide custom colors without overlapping definitions.
- for example, i can have a folder full of different colors (just like ultimaker 2+ and 3 have) without defining all these colors as separate materials. they all automatically inherit 'generic petg' settings
- this kind of scheme does not work if you make everthing non-generic material. the inheritance will be broken.
- all in all, if I want to provide a basic template within cura, 'generic' profiles seem to be a way to go

nallath, 2017-02-27:
Generic is indeed a fallback. The "inheritance" is something that's implemented, but not working a 100% (which is why we don't really use it now)

tcm0116, 2017-05-12:
> the "generic" materials were always just a sort of starting platform for your own materials rather than a material to actually be used

While that makes sense, it's a bit annoying to have a few dozen "generic" material profiles that you then have to duplicate in order to make "specific" material profiles and you can't hide or delete the "generic" profiles, so you just end up with a ton of unusable profiles.

I think it'd be great if you could edit the default profiles, but any changes you make would be revertible. Essentially, any modified change would have the reset circle arrow next to it even after you save the changes.

fieldOfView, 2017-05-29:
Cura 2.6 beta now shows a list of materials that are either (around) 1.75 or (around) 2.85 mm, depending on the material diameter specified by the printer. For a 1.75 mm printer, you don't see the UM materials, or the generic 2.85mm materials, but instead you get a list of generic 1.75 mm materials and a couple of materials from some other manufacturers.

closed 2017-11-13 by ChrisTerBeke