Ultimaker/Cura #1444: Range Slider in LayerView

fieldOfView, 2017-02-18:
I looked into unifying the two sliders in the new layerview. First I checked out the QML RangeSlider, but that does not *quite* do what I want, and it required PyQt5.8 due to an unfortunate packaging error. Instead I created one myself:
https://gist.github.com/fieldOfView/b4565ab77dbff625408bd926760ee3bf

[image]

@jackha, are you working on this or shall I continue and prepare a PR to use this in LayerView?

gabfv, 2017-02-20:
That's a great feature!

nallath, 2017-02-20:
I like it!

awhiemstra, 2017-02-20:
Nice work. As far as I know, @jackha is not currently working on any UI changes for the layer view, so feel free to make a PR.

fieldOfView, 2017-02-20:
Work in progress here:
https://github.com/Ultimaker/Cura/compare/master...fieldOfView:feature_draggable_range_slider

jackha, 2017-02-21:
Looks good! Currently I'm not working on any UI changes. Still fine-tuning the layer view itself and working on other issues.

Ghostkeeper, 2017-02-23:
Processing this and coordinating this with our UX guys and fieldOfView was just planned to be done in the sprint for the following two weeks.

fieldOfView, 2017-02-23:
I have an implementation that is mostly working, except for keyboard input. It still needs work, but that should be doable in two weeks.

fieldOfView, 2017-03-06:
Closing this issue in favor of the PR

closed 2017-03-06 by fieldOfView