Ultimaker/Cura #1445: Using external backend works only once before Cura closes the socket and kills the engine.

Neverther, 2017-02-19:
Tested on linux.
Version tested with: 2.4.0 Beta2 (Cura and CuraEngine) and with master 3bbbd4a692b41b5f85f4176fbc46cbca6511cecf CuraEngine

When using with --external-backend, the slice job is only ran once as cura closes CuraEngine and closes the listening socket.

Sockets after starting Cura (ss -lp|grep cura) shows the 127.0.0.1:49674. CuraEngine can connect to it with CuraEngine connect 127.0.0.1:49674 -j /path/to/fdmprinter.def.json -v.
After this it is shown as active connection (ss -p|grep cura).

After loading a model a slice job is ran and output from Cura shows that:
DEBUG - CuraEngineBackend.CuraEngineBackend.slice [152]: Starting slice job...
DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
DEBUG - CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [287]: Sending slice message took 0.028264760971069336 seconds
DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [360]: Slicing took 1.448084831237793 seconds
DEBUG - UM.Backend.Backend._logSocketState [175]: Socket state changed to Closing
DEBUG - UM.Backend.Backend._logSocketState [177]: Socket state changed to Closed

Cura closes the socket but does not reopen it.

And from CuraEngine:
End sliced object called. Sending 229 layers.Total time elapsed  0.68s.
Print time: 12680
Print time (readable): 3h 31m 20s
Filament: 50837
Closing connection
Arcus Error (13, native 11): We got a request to close the socket.
Arcus Error (13): Closing socket because other side requested close.

After this the CuraEngine exits and relaunching it wont connect. Socket to Cura is closed (ss does not list listening or active tcp sockets).

When launching without external backend, the connection to 127.0.0.1:49674 is active and killing the CuraEngine process is detected by Cura, it changes the socket from active to listening before relaunching CuraEngine and everything works.

The CuraEngine connect also seems to have a bug when only supplying the host without port or not supplying anything. It tries to connect to the correct port, 49674, but it does not include the supplied hostname/ip to the connection (or localhost in case of no arguments). The connection fails.

The whole thing seems kinda backwards as CuraEngine seems more like the master where Cura should be the client. Currently it is the other way around. The interface is needed before the engine, so offloading the processing to external server needs couple hacks to work.

nallath, 2017-02-20:
External backend is a debug feature which is meant to be used to run the engine yourself (so the engine can be run with gdb / valgrind / debugger).

Neverther, 2017-02-21:
Update --help with the description about it being debug feature? Or let the socket on the UI open so it becomes a functional feature?

closed 2017-02-21 by nallath