Ultimaker/Cura #1449: Can't stop moving the objects

gabfv, 2017-02-20:
Hi,

I've had a bug where I can't stop moving the object after I started to move it. Click on the left mouse button does not stop the move tool, and instead, the object simply follow the mouse.

Here are the steps to reproduce it : 

1. Open Cura.
2. Add one STL file. (It doesn't seem to matter which STL file you are choosing as I tested with different sets of files.)
3. Add another STL file.
4. Change the view to Layers.
5. Right-click on some empty space around the objects.
6. Click on Select All.
7. Click on one of the arrows to move the objects.

The move tool is now stuck. Clicking doesn't stop the move tool as it should normally do so. I've also made [a gif capture I've made of the bug](http://imgur.com/a/ksjXc). Here is [the link to the files](http://www.thingiverse.com/thing:921948) I added in the gif capture, which are _BackFootLeft.2.2.stl_ (step 2) and _BackFootRight.2.2.stl_ (step 3).

I'm using Cura 2.3.1 on Windows 10 Pro x64

nallath, 2017-02-20:
I think this was fixed in 2.4. It should be released this week.

gabfv, 2017-02-20:
Yeah, I will.

Ghostkeeper, 2017-02-23:
I couldn't reproduce this on the current Master at least.

gabfv, 2017-02-24:
Yeah, it seems fixed on 2.4 now. I tested it with different files and it seems to work as intended now.

closed 2017-02-24 by gabfv