Ultimaker/Cura #1450: "Lift head" causes head to move outside build envelope

berthrsd, 2017-02-20:
When using "Lift head" in 2.4.0 beta, the toolhead can move outside the build envelope. I found this out the hard way... :)

[image]

[Desktop.zip](https://github.com/Ultimaker/Cura/files/786946/Desktop.zip)

nallath, 2017-02-20:
I've tagged the people that did most of the work on the disallowed areas.

BagelOrb, 2017-02-20:
Ah yes. I guess someone finally found out :$

We could move toward the middle of the build plate, that shouldn't be too hard..

Note that this problem i not so much to do with disallowed areas.

Iĺl make a ticket in JIRA for it.

BagelOrb, 2017-07-21:
fixed

closed 2017-07-21 by BagelOrb