Ultimaker/Cura #1451: Cura 2.3.1 does not recognize connected Ultimaker Original+ 3D Printer

kirschnw, 2017-02-20:
When have the printer connected to my PC and start Cura 2.3.1 the printer status remains "not connected". It makes no difference, if the printer was recently connected before Cura was started or stayed connected over a restart of the system.
Looking at the Windows device list shows, that the printer is connected properly.

When I disconnect the printer and reconnect it while Cura is running, the printer will be recognozed properly. I cannot find a method to make Cura find the printer except the above described workaround

kirschnw, 2017-02-20:
The parameters of my system:
CURA version: 2.3.1
Win10 professional
USB 2.0 port

Ghostkeeper, 2017-02-23:
Does this happen in Cura 2.4 as well?

BagelOrb, 2017-07-21:
outdated

BagelOrb, 2017-07-21:
Try Cura 2.6

closed 2017-07-21 by BagelOrb