Ultimaker/Cura #1452: New material does not remain in the list when Cura 2.3.1 is restarted

kirschnw, 2017-02-20:
I defined a new material by duplicating the original one and changing some parameters in the duplicate.
The new material works fine but if Cura 2.3.1 is restarted the new material definition disappeared.
If the new definition stays activated, it remains active until I change the material but as the new material definition is no longer in the material list, I have to redefine to activate it again.

nallath, 2017-02-20:
This should be fixed in 2.4

irrenhaus, 2017-02-21:
Hey, same problem here.
I still had it with the 2.4.0-BETA2 AppImage version, just downloaded the 2.4.0 (release) AppImage version, sadly the problem still remains.
I'm on ArchLinux, KDE.

Configuration is for a Kossel Mini.

Anything I can do to test / provide necessary information?

nallath, 2017-02-22:
Create a project file with the material that you created when it's active and give us the log files.

irrenhaus, 2017-02-22:
@nallath Thanks! See attached files :)
`cura.txt` is the logfile after starting Cura, adding the material, exporting the project and closing Cura.
`cura_closed_and_restarted.txt` is the same logfile after again starting Cura (and noticing the disappeared Material definition)
`KM_RichRap_20mm_Calibration_cube_set_Part5.zip` is the project save file.

Note: I had to rename the file extensions to be able to upload them to GH.

I also noticed that when closing Cura the last message on the command line is
`./bin//cura.sh: line 11:  1073 Segmentation fault      (core dumped) cura $@`
and it does not appear in the log file.

[cura_closed_and_restarted.txt](https://github.com/Ultimaker/Cura/files/793334/cura_closed_and_restarted.txt)
[cura.txt](https://github.com/Ultimaker/Cura/files/793335/cura.txt)
[KM_RichRap_20mm_Calibration_cube_set_Part5.zip](https://github.com/Ultimaker/Cura/files/793337/KM_RichRap_20mm_Calibration_cube_set_Part5.zip)

nallath, 2017-02-22:
Hmm, this is odd. I still have the material after closing & restarting again. The kossel mini that i have a definition for doesn't even support material profiles though.

Ghostkeeper, 2017-02-23:
Nope, not fixed. This is a duplicate of https://github.com/Ultimaker/Cura/issues/1104, #1404 and #1410.

closed 2017-02-23 by Ghostkeeper