Ultimaker/Cura #1453: A bunch of issues related to adhesion and 'allowed/maximum' Print Size

filipgoc, 2017-02-20:
There are several issues I observe when slicing objects close to the max size of the build area. 

I am not sure what the underlying issue connecting all this is.

So I'll just sum up the observations with no hypothesis and see what you guys think. 

## The system set up
(Without which the observations make no sense.)
- 160x150x145
- Cura 2.4 Beta on Mac, El Capitan
- Custom Marlin based printer (IMADE3D JellyBOX)
[image]

## Interaction between X and Y
- x=150, y=150, slices fine
- X=160, y=100, slices fine
- X=160, y=150, **error**
- same pattern works in the other direction:
- y=160, x=100, slices fine
- y=160, x=150, **error**
[image]
[image]

## Can't reach 'full size' no matter what
- Adhesion is set to 'none'
- maximum sizes in x and y are 12mm and 9mm smaller than the printer print area
[image]

## Print area reduced even if adhesion not enabled
- adhesion not enabled
- yet, as clear from the dark grey border, there is some ares taken away from maximum XY size
- when we enable adhesion, it adds a 2nd grey border
- it therefore seems the first one is _not_ for adhesion... so what is it? 
[image]
[image]

nallath, 2017-02-20:
We add a very small area to the shadow an object "drops" to prevent rounding errors in the model. If I remember correctly it's a 0.5 mm offset, which can contribute to a few of the issues you observe. 

Adhesion is not the only setting that has influence on the build size. A number of travel & avoid settings also influence it (infill_wipe_dist, travel_avoid_distance, travel_avoid_other_parts)

awhiemstra, 2017-02-20:
Note also that the recommended mode "build plate adhesion" toggle toggles between brim and skirt, not between brim and none.

filipgoc, 2017-02-21:
Not all sure the tiny buffer you mention (0.5mm) would result in 
> x and y are 12mm and 9mm smaller than the printer print area

I'll explore the infill_wipe_dist, travel_avoid_distance, travel_avoid_other_parts and see what happens.

Ghostkeeper, 2017-02-23:
The full list of settings that influence the disallowed areas are here: https://github.com/Ultimaker/Cura/blob/b0c8a4df7a995520665e701f2f829a69ebbfba14/cura/BuildVolume.py#L891

The `_extruder_settings` only affect which extruders are used (for multi-extrusion machines) and the `_extra_z_settings` only affect the height of the build volume.

To use the full build volume, you typically need to disable platform adhesion and disable Avoid Objects While Printing. Other things that affect the build volume (Draft Shield, Ooze Shield and Infill Wipe Distance) are typically already disabled by default, unless you're using some machine definition I don't typically test with.

filipgoc, 2017-02-26:
thanks for this @Ghostkeeper . 
i'll report back after another round of test :-)

ianpaschal, 2018-02-21:
This came up again in [#3365](https://github.com/Ultimaker/Cura/issues/3365) including some discussion of possible fixes/improvements, so I'm closing this. Discussion can be continued in the up-to-date thread.

closed 2018-02-21 by ianpaschal