Ultimaker/Cura #1455: Project saving issues (multiplying extensions, no keyboard support)

filipgoc, 2017-02-20:
Mac El Capitan, Cura 2.4 Beta

## no keyboard
When saving a project (summary showing), pressing `enter` simply closes the window instead of confirming the `save` button. One has to either use the mouse (oh the pain) or press tab a bunch of times and then use the spacebar. 
[image]

## .curaproject.3fm not generated properly; cyclical error
Even though the (automatically generated!) filename `3Dbenchy.curaproject.3mf` looks ok, cura screams:
[image]

Consequently, file is saved as `3DBenchy.curaproject.curaproject.3mf`
Moreover, if I save the project again (save as), the automatically generated filename is `3DBenchy.curaproject.3mf.curaproject.3mf` and cura still screams
[image]

Weird?

nallath, 2017-02-21:
It's not Cura that screams, it's OSX that screams. It works fine on Linux & Windows. Unfortunately it's one of those issues that is out of our hands :(

filipgoc, 2017-02-21:
Ha. That's good to know. 

In that case I post here: 

## Mac OS : Saving .curaproject.3mf Workaround/ Deal with It
- As long as you're saving a file with no visible extension, it will be A-OK. 
- When you're saving a project, the filename is automatically highlighted, including the extension. 
- Therefore, if you simply rename the file by starting to type with no mouse input, and then save the project, all will be well. And it's not that much work. 

[image]
[image]

filipgoc, 2017-02-21:
@nallath and the issue of not saving on 'enter' ?
Is that out of hands as well?

nallath, 2017-02-21:
Uh, I'm not sure about that one.

DDDirk, 2017-02-21:
More in general, Cura could be a lot more keyboard-friendly. That would not only speed up work, but also help prevent RSI.

filipgoc, 2017-02-21:
Oh I dare not even to mention that @DDDirk 
I cry salty tears every time I don't use a keyboard shortcut!

I have defined many shortcuts myself (export to file, open project, save project... ), but not all can be easily defined. 

And even the ones I have (mac has a native way to do it), but honestly even these shortcuts don't work every time, they are finicky in cura (even though for other programs I use this approach with no issues). 

Another thread on shortcuts?!
I am happy to make a the first draft of the desired ones.

nallath, 2017-02-21:
Adding short cuts should not be that hard for most of them. A number of actions (save file, etc) we use default shortcuts.

fieldOfView, 2017-02-21:
There's a JIRA issue for improved keyboard navigation (and another one for improved touch support btw), but I am guessing that has been postponed to kingdom come.

filipgoc, 2017-02-21:
I've never implemented keyboard shortcuts into a project from developer's side, but if there's some helper type work (give me a schema and I'll can prepare the bindings map for example), happy to help.

nallath, 2017-02-22:
We use the QT shortcuts. For the moment being it means that the key sequences defined in http://doc.qt.io/qt-5/qkeysequence.html#StandardKey-enum would work. It's also possible to add your own, but thats a bit more work.

Ghostkeeper, 2017-02-23:
I think this particular dialogue is not a problem of introducing keyboard shortcuts, but a problem of setting the focus on the correct button initially.

DDDirk, 2017-03-04:
Related is using the keyboard for navigation through the menus. The names have underscores for Alt-key-combinations (visible without pressing Alt first, which I like). But (in Cura 2.4) these don't work for View and Settings. I can acces them through Alt-E (edit) and then the right arrow. But when I do that 3 times I should get into the printer submenu, which doesn't happen (although it becomes visible, but not selected (highlighted)). If I use the right arrow a fourth time, the menus disappear. This also happens when the submenu is visible, eg File > Open recent. Or nothing happens, as with the submenus under Extensions (meaning I can't move on to Preferences that way).

It really disrupts my workflow if things don't work as expected. I have to momentarily think about something that should be automatic and then I have to grab the mouse, which also slows me down. And it's bad for my RSI.

jackha, 2018-07-30:
Pressing enter will now open the file dialog to save: c036912

closed 2018-07-30 by jackha