Ultimaker/Cura #1456: Autosave Current Workspace as Current-Project on Cura Quit (Feature Idea)

filipgoc, 2017-02-20:
Now that Cura supports Projects, I propose that when Cura quits, the “workspace”* get automatically saved as a Current-Project in the user’s folder, and gets loaded next time Cura starts instead of “blank slate.”

- By workspace I mean models with their transformations and slicing settings.
- May be beneficial to have this as a turn off/on option in preferences.

nallath, 2017-02-21:
Right now, workspace saving can be a bit on the expensive side. Especially if you have a build plate filled with models, it can take ~10 seconds to save it. I do like the idea, but im not so sure about the user impact.

Ghostkeeper, 2017-06-05:
Our product manager just closed this for being inactive for 3 months.

nickthetait, 2017-06-07:
Has any work been done to the performance of saving a workspace? This may be less of a hurdle nowadays...

awhiemstra, 2017-06-08:
Some work has been done, we introduced https://github.com/ultimaker/libSavitar to speed up the 3MF reading/writing. Parts of workspace are still being done in Python though, so there are still performance bottlenecks. We might be able to do project save on window close though, but we definitely would not want to do project auto saves while the user is using the application.

nallath, 2017-06-08:
The ~10 sec is the current time for larger workspaces. Autosaving is not that good of an idea at the moment (unless we put a whole lot of work in it).

closed 2018-02-27 by ianpaschal