Ultimaker/Cura #1457: UI Post Processing Extension Window too Big (Bug?) on Mac

filipgoc, 2017-02-20:
Incorrect sizing of Modify G-code post processing extension. 
- mac el capitan cura 2.4 beta
- The presumed `OK` button is out of display and invisible. 
- There is no visual way to interact, or close the window. One just has to try “enter” and hope for the best. 
[image]

fieldOfView, 2017-02-20:
Hmm, weird. Is this just with the post processing window? Or are windows such as the About dialog and Preference window also affected?

filipgoc, 2017-02-21:
-- | --
-- | --
preferences    | ok
About dialog   | oversized
changelog       | ok

nallath, 2017-02-21:
Ugh. Stupid DPI densitity issues again. I thought we squashed all of them :(

nallath, 2017-07-27:
This should be fixed by the pull request!

closed 2017-07-27 by nallath