Ultimaker/Cura #1458: 2.4.0b2 crashes at VersionUpgrade22to24

wkearney99, 2017-02-20:
Seems like something's going wrong detecting a USB-connected printer that's on COM3.

The code is crashing:

2017-02-20 18:03:31,846 - WARNING - UM.Logger.logException [65]: Exception: Exception in machine_instance upgrade with VersionUpgrade22to24.VersionUpgrade: Traceback (most recent call last):
  File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\VersionUpgradeManager.py", line 310, in _upgradeFile
TypeError: 'NoneType' object is not iterable

What's it looking for there that's causing it to throw the exception?

nallath, 2017-02-21:
It's trying to upgrade a file to the next version. As far as I can see, this should be fixed already. It definately has nothing to do with USB.

Ghostkeeper, 2017-02-23:
Something went wrong in upgrading old files from your 2.3 installation to the format we use for 2.4. The update function returned `None`, while it should return a tuple (whose items may be `None`).

Ghostkeeper, 2017-02-23:
Not sure how we can debug this. Perhaps you could zip up your entire Cura configuration folder (`C:\Users\wkearney99\AppData\Local\cura`) and attach it to this issue?

fieldOfView, 2017-02-23:
This would have prevented the crash, but it was not applied to the 2.4 branch:
https://github.com/Ultimaker/Uranium/pull/209/files

closed 2017-11-13 by ChrisTerBeke