Ultimaker/Cura #1459: Cura 2.3.1 not showing model in layer view

TheUnHumanZone, 2017-02-21:
I've had this problem a long time and don't know how to fix it.

fieldOfView, 2017-02-21:
2.4 has just been released. It might fix your issue. If not, we need your logs.

MatteoRagni, 2017-08-27:
I'm experiencing the same problem.

[the log](https://gist.github.com/MatteoRagni/7b0c286435b397ffad9e4bd541043ff8)

and an image of what is happening:

If all layers are selected to be visible everything is ok:
[image]
but if I remove even one layer:
[image]

Ghostkeeper, 2017-09-02:
This doesn't reproduce on my end.

ChrisTerBeke, 2018-01-22:
Unable to reproduce and no comments for almost a year. Closing until re-opened.

closed 2018-01-22 by ChrisTerBeke