Ultimaker/Cura #1460: Cura 2.4 could not save gcode / Windows 10

frangomezol, 2017-02-21:
When I try to save the gcode after slicing the part I get this message, and the gcode is not writed to the disk.
Message: Could not save to x:/UM3_GYRO_FGO.gcode: 'NoneType' objet has no attribute 'getMetaDataEntry'

[image]

nallath, 2017-02-22:
Could you provide us with the logs please?

frangomezol, 2017-02-22:
Yes, hereby you will find stderr.log, cura log.

[cura.zip](https://github.com/Ultimaker/Cura/files/793178/cura.zip)

Ghostkeeper, 2017-02-23:
Not sure why you closed it, it seems like a genuine issue.

This is the stack trace:

    2017-02-22 10:26:46,242 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [138]: Writing to Local File Z:/A Clientes Piezas/Aerotools Jose Luis Alvarez/20170220 incendio/UM3_Soporte camera_socket 2.gcode in text mode
    2017-02-22 10:26:47,591 - ERROR - UM.Logger.logException [65]: Exception: Job <UM.FileHandler.WriteFileJob.WriteFileJob object at 0x000001712CFD8FD0> caused an exception
    2017-02-22 10:26:47,599 - ERROR - UM.Logger.logException [69]: Traceback (most recent call last):
    2017-02-22 10:26:47,602 - ERROR - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 122, in run
    2017-02-22 10:26:47,606 - ERROR - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\FileHandler\WriteFileJob.py", line 37, in run
    2017-02-22 10:26:47,609 - ERROR - UM.Logger.logException [69]:   File "D:\Program Files\Cura 2.4\plugins\GCodeWriter\GCodeWriter.py", line 71, in write
    2017-02-22 10:26:47,613 - ERROR - UM.Logger.logException [69]:     settings = self._serialiseSettings(Application.getInstance().getGlobalContainerStack())
    2017-02-22 10:26:47,617 - ERROR - UM.Logger.logException [69]:   File "D:\Program Files\Cura 2.4\plugins\GCodeWriter\GCodeWriter.py", line 133, in _serialiseSettings
    2017-02-22 10:26:47,620 - ERROR - UM.Logger.logException [69]:     flat_extruder_quality.addMetaDataEntry("quality_type", extruder.findContainer({"type": "quality"}).getMetaDataEntry("quality_type", "normal"))
    2017-02-22 10:26:47,623 - ERROR - UM.Logger.logException [69]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

Ghostkeeper, 2017-02-23:
It seems like somehow there was no active quality profile in one of the extruders.

Ghostkeeper, 2017-02-23:
No, actually, it seems to be caused by a faulty upgrade again of the settings files from version 2.3 to 2.4.

frangomezol, 2017-02-23:
Sorry I did not want to close it, I am new at github...
Please keep it open!!
Thanks

frangomezol, 2017-02-23:
Aditional info: after the error, if I skip it, the gcode file is created in the USB. So just now i can work.
regards

ianpaschal, 2018-02-21:
Since this is more than a year old and we are not going to work on 2.x anymore, I'm closing this issue.

closed 2018-02-21 by ianpaschal