Ultimaker/Cura #1461: Cura 2.4 - add 1.0mm nozzle option to UM2+

MarcusWolschon, 2017-02-21:
I can not create an Ultimaker II with Olsson Block.
Using the Ultimaker II+ profile instead, I am limited to only 0.25, 0.4, 0.6 and 0.8mm nozzles.
However I do all structural parts with a 1.0mm nozzle that can not be selected.
(very strong parts in a short time where details don't matter)

ChrisTerBeke, 2018-01-22:
As a work-around you can edit the line width instead of the nozzle size.

Ghostkeeper, 2018-02-09:
We may not have specific profiles for those nozzle sizes, but you can still modify the line width yourself if you modified the hardware of the printer.

ChrisTerBeke, 2018-03-02:
Closing due to old Cura version and the fact that we won't support custom nozzle sizes on Ultimaker printers (not testable for us).

closed 2018-03-02 by ChrisTerBeke