Ultimaker/Cura #1462: Spiral mode does not work properly

DmitryBychkov, 2017-02-21:
using this model http://www.thingiverse.com/thing:1931480     Watering_can.stl
scaled 2x1.2x1.2
I get instead of a spiral separate layers with holes between them.
Example:
G1 X147.069 Y90.733 Z8.2 E4638.44268
;TIME_ELAPSED:1255.436092
;LAYER:20
G0 F1800 X147.05 Y90.648
;TYPE:WALL-OUTER
G1 F1200 X148.574 Y89.66 Z8.202 E4638.74473

This bug was in the version 13.04,14.12.1 and nothing changes in 2.3:
z seam alignment shortest:
[image]
z seam alignment back:
[image]

I tried to make the post-processing of G-code. 
I commented out G0 and add G92 E command
example:
G1 X147.069 Y90.733 Z8.2 E4638.4426
;TIME_ELAPSED:1255.436313
;LAYER:20
;G0 F1800 X147.05 Y90.648
;TYPE:WALL-OUTER
G92 E4638.43726
G1 F1200 X148.574 Y89.66 Z8.202 E4638.74464

much better:
[image]

but not perfect...

smartavionics, 2017-02-22:
Hi folks, I know what the problem is here and am currently testing a solution. Stay tuned...

smartavionics, 2017-03-08:
Hi Dimitry, cura 2.5 should do a much better job as the spiralization code has been revamped.

BagelOrb, 2017-03-08:
Major props to @smartavionics for that!

closed 2017-03-08 by BagelOrb