Ultimaker/Cura #1463: Ultimaker 2 USB printing

softtouch, 2017-02-22:
I plugged in my ultimaker 2 via usb, and cura detected it and showed the button to print via usb.
Then when I actually wanted to print, it told me that the printer does not support usb printing.
So why does it show the button in the first place after detecting the printer?

fieldOfView, 2017-02-22:
The Ultimaker 2 and 2+ do not officially support USB printing. You can make it work by going to the "Printers" pane of the preferences, selecting your UM2 and going into the Machine Settings. Select "RepRap (Marlin/Sprinter)" as the gcode flavor.

Ghostkeeper, 2017-02-24:
The reason why it is not supported is because the UM2 is supposed to regulate its own temperature based on the material currently selected. It does regulate that properly, but when printing via USB it doesn't do that, it simply executes the g-codes coming in via USB blindly.

When generating g-code for the UM2, we don't include any temperatures because the UM2 normally regulates the temperatures itself. When you then decide to print via USB it won't heat up the nozzles.

It should've been fixed by prefixing some temperature commands when printing via USB. However that would have it show like we actually support USB printing with the Ultimaker 2 and it's been decided that we shouldn't spend the resources on that. Switching the flavour to RepRap works fine.

closed 2017-02-24 by Ghostkeeper