Ultimaker/Cura #1464: Retraction points in layer view not visible

Dunkc, 2017-02-22:
Hi
I believe that the retraction points should show up in the Layers View as vertical blue lines.  I am using 2.3.1 on a Mac but they are not showing.  Retraction is enable in the toolbar with 4 mm height.
[image]

nallath, 2017-02-22:
@jackha Did we change this for the improved layerview?

Ghostkeeper, 2017-02-24:
No, we didn't. Lines are only horizontal. Retractions are visualised by making the following travel move steel-blue instead of dark blue.

There's an issue open to visualise 3D moves but it's been put in the we're-not-really-doing-this sprint. This issue would make things like Z-hops and wire printing visible.

tonycstech, 2019-10-25:
Great. Take away useful information and introduce hundreds of repeating settings.
Sure new CURA is better then ever.

I have to use external program to view how printer will behave before i print.
Isnt that what the layer preview should be for ?
It should shows me every move the nozzle makes and give me information about what its actually doing.
So far only Mattercontrol and Craftware are doing it.