Ultimaker/Cura #1465: Infill Values not working

KacieHultgren, 2017-02-22:
Changing infill values in Cura 2.4 on mac is not working.  Have tried changing via the recommended tab - see attached images - as well as custom tab.  Changes in infill pattern also not working.

[image]
[image]
[image]

awhiemstra, 2017-02-22:
It is immediately obvious here but you are changing the infill density of the second extruder, while your object is probably being printed with the first extruder. If you switch to the first extruder (click on the "Print Core 1" tab thingy) you should see the infill density properly change.

KacieHultgren, 2017-02-22:
Ah, thank you.  I haven't used the extruder often enough to get the tab switching into my workflow.  Thanks for catching.

closed 2017-02-22 by nallath