Ultimaker/Cura #1466: Strange behavior after switching Nylon to PLA to Nylon changing protected profile changed Extruder temp to 160C

partsoven, 2017-02-22:
The other day I swapped from Nylon to PLA back to Nylon. All of a sudden my printer jammed up and appeared to be stripping/grinding. I thought it was a feeder issue first so I disassembled to the feeder... twice, swapped the AA nozzle, contacted support, and ran through the troubleshooting.

Then after scratching my head all day, I finally noticed the temp in the protected Ultimaker profile for Nylon was all of a sudden set to 160C for some reason. No wonder it wasn't extruding. 

I never edited this value. This only happened when I went Nylon to PLA to Nylon. I am only using Ultimaker materials on this machine with the protected Ultimaker profiles so it appears to be a bug to me.

Ghostkeeper, 2017-02-24:
Hmm, can't seem to reproduce this. I switched the left extruder to Black Nylon, then switched it to Silver Metallic PLA, then back to Black Nylon. The printing temperature ends up at 245 degrees, which seems fine.

Perhaps you could show us a log? Reproduce the bug in Cura, then upload a copy of stderr.log here?

partsoven, 2017-02-27:
Where is that log? I'd be happy to upload.

Ghostkeeper, 2017-03-01:
Depends on which operating system you're using. See https://github.com/Ultimaker/Cura#logging-issues

closed 2018-02-27 by ianpaschal