Ultimaker/Cura #1467: i cant open files

Albert7758, 2017-02-22:


fieldOfView, 2017-02-22:
We need more details. What version of Cura on what OS? How do you try to open files? Got logs?

Ghostkeeper, 2017-02-24:
Can't reproduce this, files open fine. We'll need more specifics.

fieldOfView, 2017-02-24:
I *think* this might be the bug where on the first launch on Windows, you cannot drag and drop files into Cura. This is happens only once (when running Cura from the installer) and is automagically solved by restarting Cura.

That, or @Albert7758 has glued his folders shut with the glue-stick that came with his printer and he cannot access the files inside them. I can't tell.

Ghostkeeper, 2017-02-24:
Just the other day I found an old binder with some files under my bed, with some physics exercises from high school. The binder with files opened fine but it was a little dusty. If he used the glue stick that might've been a problem.

fieldOfView, 2017-02-24:
@Albert7758, we're not making fun of you (well, ok, a little). Please let us know if my first suggestion was indeed what you meant, or if you have found another issue.

closed 2017-07-27 by nallath