Ultimaker/Cura #1468: Ultimaker 1 stops waiting for non-existent heated bed when printing multiple models

PaulMurrin, 2017-02-22:
Using Cura 2.4.0, I'm slicing a part which I've replicated multiple times.
I'm using an ultimaker original with no heated bed and it is configured correctly in the printer settings. When the printer has completed the final layer of the 1st part the printer stops and displays a message that it is waiting for heated bed. The only way to release this is via power cycle.
The same part sliced fine with version 2.3.0 (but not with 2.4.0 beta).

Ghostkeeper, 2017-02-24:
I could reproduce this. Slicing a small cube for UMO has the following output around layer 1:

    G0 F3600 X103.5 Y101
    ;TIME_ELAPSED:40.123234
    ;LAYER:1
    M140 S60 ;<-- This one is wrong!
    M106 S255
    G0 F5400 X103.9 Y100.6 Z0.4
    ;TYPE:WALL-INNER
    G1 F900 X103.9 Y104.4 E17.19477

This seems to be caused by implementing the first layer bed temperature. It skips the bed temperature initially when you don't have a heated bed, but when it reaches the second layer there is no such check. We should introduce it. Should be a simple fix.

fieldOfView, 2017-02-24:
A temporary fix could be to lie to Cura that your UMO has a heated bed, and setting the bed temperature to 0 (I have not tested this)

Ghostkeeper, 2017-02-24:
Made a fix here: https://github.com/Ultimaker/CuraEngine/commit/c412a6ccfed9526fa09cf57f2dac6c44172a88ea

Seems to work properly now.

closed 2017-03-08 by BagelOrb