Ultimaker/Cura #1470: No z value in gcode

Handieman2, 2017-02-22:
Hello 

I'm trying to use cura on my kossel delta printer. I've tried different settings but in none of them does cura provide a z value for every layer. Causing my printer to start at the first layer height and staying there for the rest of the print. Does anyone know how to fix this?
Example: At none of the G0 commands does it tell to go to a different Z value
;FLAVOR:RepRap
;TIME:895
;Filament used: 1.05575m
;Layer height: 0.3
;Generated with Cura_SteamEngine 2.4.0
M190 S60
M104 S205
M109 S205
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 ;Home all axes (max endstops)
G29 ; Bed Leveling
G1 Z0.3 F9000 ;move the platform down 0.3mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
M117 Printing...
;LAYER_COUNT:66
;LAYER:0
M107
G1 F1500 E-6.5
G0 F3600 X21.8 Y-10 Z0.15
;TYPE:SKIRT
G1 F1500 E0
G1 F1800 X21.8 Y10 E0.37418
G1 X21.76 Y10.971 E0.39236
G1 X21.64 Y11.935 E0.41053
G1 X21.442 Y12.885 E0.42869
G1 X21.166 Y13.817 E0.44687
G1 X20.814 Y14.722 E0.46504
G1 X20.389 Y15.596 E0.48322
G1 X19.893 Y16.431 E0.50139
G1 X19.331 Y17.223 E0.51956
G1 X18.705 Y17.966 E0.53774
G1 X18.02 Y18.655 E0.55592
G1 X17.281 Y19.286 E0.5741
G1 X16.493 Y19.853 E0.59226
G1 X15.661 Y20.354 E0.61043
G1 X14.79 Y20.784 E0.6286
G1 X13.886 Y21.142 E0.64679
G1 X12.957 Y21.424 E0.66495
G1 X12.007 Y21.628 E0.68313
G1 X11.044 Y21.754 E0.7013
G1 X10 Y21.8 E0.72085
G1 X-10 Y21.8 E1.09503
G1 X-10.971 Y21.76 E1.11321
G1 X-11.935 Y21.64 E1.13139
G1 X-12.885 Y21.442 E1.14954
G1 X-13.817 Y21.166 E1.16773
G1 X-14.722 Y20.814 E1.18589
G1 X-15.596 Y20.389 E1.20408
G1 X-16.431 Y19.893 E1.22225
G1 X-17.223 Y19.331 E1.24042
G1 X-17.966 Y18.705 E1.25859
G1 X-18.655 Y18.02 E1.27677
G1 X-19.286 Y17.281 E1.29495
G1 X-19.853 Y16.493 E1.31311
G1 X-20.354 Y15.661 E1.33128
G1 X-20.784 Y14.79 E1.34946
G1 X-21.142 Y13.886 E1.36765
G1 X-21.424 Y12.957 E1.38581
G1 X-21.628 Y12.007 E1.40399
G1 X-21.754 Y11.044 E1.42216
G1 X-21.8 Y10 E1.44171
G1 X-21.8 Y-10 E1.81589
G1 X-21.76 Y-10.971 E1.83407
G1 X-21.64 Y-11.935 E1.85224
G1 X-21.442 Y-12.885 E1.8704
G1 X-21.166 Y-13.817 E1.88858
G1 X-20.814 Y-14.722 E1.90675
G1 X-20.389 Y-15.596 E1.92493
G1 X-19.893 Y-16.431 E1.9431
G1 X-19.331 Y-17.223 E1.96127
G1 X-18.705 Y-17.966 E1.97945
G1 X-18.02 Y-18.655 E1.99762
G1 X-17.281 Y-19.286 E2.0158
G1 X-16.493 Y-19.853 E2.03397
G1 X-15.661 Y-20.354 E2.05214
G1 X-14.79 Y-20.784 E2.07031
G1 X-13.886 Y-21.142 E2.0885
G1 X-12.957 Y-21.424 E2.10666
G1 X-12.007 Y-21.628 E2.12484
G1 X-11.044 Y-21.754 E2.14301
G1 X-10 Y-21.8 E2.16256
G1 X10 Y-21.8 E2.53674
G1 X10.971 Y-21.76 E2.55492
G1 X11.935 Y-21.64 E2.5731
G1 X12.885 Y-21.442 E2.59125
G1 X13.817 Y-21.166 E2.60944
G1 X14.722 Y-20.814 E2.6276
G1 X15.596 Y-20.389 E2.64579
G1 X16.431 Y-19.893 E2.66396
G1 X17.223 Y-19.331 E2.68213
G1 X17.966 Y-18.705 E2.7003
G1 X18.655 Y-18.02 E2.71848
G1 X19.286 Y-17.281 E2.73666
G1 X19.853 Y-16.493 E2.75482
G1 X20.354 Y-15.661 E2.77299
G1 X20.784 Y-14.79 E2.79116
G1 X21.142 Y-13.886 E2.80936
G1 X21.424 Y-12.957 E2.82752
G1 X21.628 Y-12.007 E2.8457
G1 X21.754 Y-11.044 E2.86387
G1 X21.8 Y-10 E2.88342
G0 F3600 X21.4 Y-10
G1 F1800 X21.4 Y10 E3.25759
G1 X21.36 Y10.954 E3.27546
G1 X21.24 Y11.901 E3.29332
G1 X21.042 Y12.835 E3.31118
G1 X20.766 Y13.749 E3.32904
G1 X20.414 Y14.637 E3.34691
G1 X19.99 Y15.492 E3.36477
G1 X19.495 Y16.309 E3.38264
G1 X18.934 Y17.081 E3.40049
G1 X18.31 Y17.804 E3.41836
G1 X17.628 Y18.472 E3.43622
G1 X16.892 Y19.081 E3.45409
G1 X16.108 Y19.625 E3.47195
G1 X15.281 Y20.103 E3.48982
G1 X14.418 Y20.509 E3.50766
G1 X13.523 Y20.842 E3.52553
G1 X12.603 Y21.099 E3.5434
G1 X11.665 Y21.278 E3.56126
G1 X10.716 Y21.378 E3.57912
G1 X10 Y21.4 E3.59252
G1 X-10 Y21.4 E3.9667
G1 X-10.954 Y21.36 E3.98456
G1 X-11.901 Y21.24 E4.00242
G1 X-12.835 Y21.042 E4.02028
G1 X-13.749 Y20.766 E4.03814
G1 X-14.637 Y20.414 E4.05601
G1 X-15.492 Y19.99 E4.07387
G1 X-16.309 Y19.495 E4.09174
G1 X-17.081 Y18.934 E4.10959
G1 X-17.804 Y18.31 E4.12746
G1 X-18.472 Y17.628 E4.14532
G1 X-19.081 Y16.892 E4.16319
G1 X-19.625 Y16.108 E4.18105
G1 X-20.103 Y15.281 E4.19892
G1 X-20.509 Y14.418 E4.21676
G1 X-20.842 Y13.523 E4.23463
G1 X-21.099 Y12.603 E4.2525
G1 X-21.278 Y11.665 E4.27036
G1 X-21.378 Y10.716 E4.28822
G1 X-21.4 Y10 E4.30162
G1 X-21.4 Y-10 E4.6758
G1 X-21.36 Y-10.954 E4.69366
G1 X-21.24 Y-11.901 E4.71152
G1 X-21.042 Y-12.835 E4.72938
G1 X-20.766 Y-13.749 E4.74724
G1 X-20.414 Y-14.637 E4.76511
G1 X-19.99 Y-15.492 E4.78297
G1 X-19.495 Y-16.309 E4.80084
G1 X-18.934 Y-17.081 E4.8187
G1 X-18.31 Y-17.804 E4.83656
G1 X-17.628 Y-18.472 E4.85442
G1 X-16.892 Y-19.081 E4.8723
G1 X-16.108 Y-19.625 E4.89015
G1 X-15.281 Y-20.103 E4.90802
G1 X-14.418 Y-20.509 E4.92586
G1 X-13.523 Y-20.842 E4.94373
G1 X-12.603 Y-21.099 E4.9616
G1 X-11.665 Y-21.278 E4.97946
G1 X-10.716 Y-21.378 E4.99732
G1 X-10 Y-21.4 E5.01072
G1 X10 Y-21.4 E5.3849
G1 X10.954 Y-21.36 E5.40276
G1 X11.901 Y-21.24 E5.42062
G1 X12.835 Y-21.042 E5.43848
G1 X13.749 Y-20.766 E5.45634
G1 X14.637 Y-20.414 E5.47421
G1 X15.492 Y-19.99 E5.49207
G1 X16.309 Y-19.495 E5.50994
G1 X17.081 Y-18.934 E5.5278
G1 X17.804 Y-18.31 E5.54566
G1 X18.472 Y-17.628 E5.56352
G1 X19.081 Y-16.892 E5.5814
G1 X19.625 Y-16.108 E5.59925
G1 X20.103 Y-15.281 E5.61712
G1 X20.509 Y-14.418 E5.63496
G1 X20.842 Y-13.523 E5.65283
G1 X21.099 Y-12.603 E5.6707
G1 X21.278 Y-11.665 E5.68857
G1 X21.378 Y-10.716 E5.70642
G1 X21.4 Y-10 E5.71982
G0 F3600 X21 Y-10
G1 F1800 X21 Y10 E6.094
G1 X20.96 Y10.937 E6.11154
G1 X20.84 Y11.867 E6.12909
G1 X20.642 Y12.784 E6.14664
G1 X20.366 Y13.68 E6.16418
G1 X20.015 Y14.55 E6.18173

fieldOfView, 2017-02-23:
Cura does not create z coordinates if the z does not change. Since all the moves in your snippet are from the first layer they all have the same z coordinate, so it is included only once (the G0 after ;LAYER: 0). In normal slice results, each layer has a single Z coordinate.

Handieman2, 2017-02-23:
Thanks for explaining. I found the problem on my printer. z=0 was not set at the right height. Meaning the first layer was lower then 0 (negative)

closed 2017-02-23 by Handieman2